carla-simulator/carla #6486: Unreal Setup.bat (failed to download error)

pjk19942, 2023-05-20:


pjk19942, 2023-05-20:
Hi, I am new to CARLA and trying to install via the Windows Build. In the step where we need to install UnrealEngine, and run the Setup.bat , I get the following error. 

Checking dependencies...
Updating dependencies:   0% (0/63485)...
Failed to download 'http://cdn.unrealengine.com/dependencies/UnrealEngine-12372779-e1515af26c634d2a8ade60b1afd1f065/01bb78539fc8dda386d45f9b5615f9a1e8ca5d94': The remote server returned an error: (403) Forbidden. (WebException)

This issue was later submitted on Unreal Forum due to some maintenance which they are trying to fix/or maybe is fixed
https://forums.unrealengine.com/t/upcoming-disruption-of-service-impacting-unreal-engine-users-on-github/1155880

The update was that they added was :
"_To remedy the error in the builds, a new Commit.gitdeps.xml file has been attached to each release version as an asset. Please replace the existing Engine/Build/Commit.gitdeps.xml file with the new asset._"

Now in order to use CARLA, do I need to simply download via: https://github.com/EpicGames/UnrealEngine/tree/4.26
or will there be any updates made to the:
    git clone --depth 1 -b carla https://github.com/CarlaUnreal/UnrealEngine.git .

Thanks
Prakhar

pjk19942, 2023-05-22:
Just saw the post on https://github.com/carla-simulator/carla/issues/6454 . Redownloading the Engine/Build/Commit.gitdeps.xml from https://github.com/EpicGames/UnrealEngine/tree/4.26/Engine/Build fixed it. :)

closed 2023-05-22 by pjk19942

jaifien, 2023-08-29:
@pjk19942 Fixed the issue for me

pravin-appa, 2024-09-26:
Failed to download 'http://cdn.unrealengine.com/dependencies/UnrealEngine-17152897/03a48e95a11f19cb8c0c2433929369fb8ef5ab4c': The remote server returned an error: (500) Internal Server Error. (WebException)

this error is still showing even after replacing XML file

InS1D3By73, 2025-02-23:
Could someone tell me why, when I run setup.bat, I am able to download all dependencies until I encounter this error:

'https://cdn.unrealengine.com/dependencies/UnrealEngine-35833348/1cc360b336bd01171e2418d83812d15fc5f0ce24': Request canceled: Unable to create a secure SSL/TLS channel. (WebException)