carla-simulator/carla #6487: New vehicle sinks into the ground

emanef13, 2023-05-21:
Hello CARLA community,

I have imported a new vehicle to CARLA.
I have followed the documentation on how to import a new vehicle in CARLA, however when spawning the new car in my own map, it just sinks half below the ground half above the ground.
I have tried changing the wheel radius blueprint, also setting the shape to UsePhysAssetShape, but nothing seems to be solving the issue.
Any help would be very helpful, thanks :)

ivevasiljevic, 2023-05-23:
Have you checked and changed the vehicle's Transform attribute, or more precisely it's location at the z coordinate (https://carla.readthedocs.io/en/latest/python_api/#carla.Transform). Does the car spawn normally, and then when it hits the ground, it just sinks?

svenskiy, 2023-07-26:
Same issue.
I exported Tesla model from Unreal Engine to Blender as a template and add new car, then binded bones and export model to fbx. Then i passed all steps according instruction and can spawn my vehicle into world, but my tires stuck in road texture. I tried to change physical vehicle body in  Physics Asset to raise the model above road, but after spawn car stuck in air. 

[image]

Asapling, 2024-02-18:
If the z-axis in the generated vehicle's position coordinates is above the road surface, will the vehicle model fall and get stuck on the road surface?

closed 2025-04-07 by Blyron