carla-simulator/carla #6492: 'Cast' errors: CarlaSettingsDelegate.cpp (lines 161-162)

4doschris, 2023-05-22:
Hi all!

I have a really hard time figuring out what could be the root cause of these errors. I have not found a SINGLE report on this problem.

CARLA version: 0.9.14.
Platform/OS: Windows 11

After embedding a custom radar alogrithm, this bunch of error messages crossed my path again and again. I have rebuilt Carla from source, when I first bumped into it, but ever since I have experienced the same problem again.

**Here are the error codes:**

C:/FETI/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Settings/CarlaSettingsDelegate.cpp(161): error C2065: 'UCarlaGameInstance': undeclared identifier
C:/FETI/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Settings/CarlaSettingsDelegate.cpp(161): error C2672: 'Cast': no matching overloaded function found
C:/FETI/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Settings/CarlaSettingsDelegate.cpp(161): error C2974: 'Cast': invalid template argument for 'T', type expected
C:\FETI\UnrealEngine\Engine\Source\Runtime\Engine\Classes\Engine/World.h(105): note: see declaration of 'Cast'
C:/FETI/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Settings/CarlaSettingsDelegate.cpp(161): error C2974: 'Cast': invalid template argument for 'To', type expected
C:\FETI\UnrealEngine\Engine\Source\Runtime\CoreUObject\Public\Templates/Casts.h(109): note: see declaration of 'Cast'
C:/FETI/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Settings/CarlaSettingsDelegate.cpp(162): error C3536: 'GameInstance': cannot be used before it is initialized
C:/FETI/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Settings/CarlaSettingsDelegate.cpp(162): error C2446: '!=': no conversion from 'nullptr' to 'int'
C:/FETI/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Settings/CarlaSettingsDelegate.cpp(162): note: A native nullptr can only be converted to bool or, using reinterpret_cast, to an integral type

And here is the problematic code part where the error codes point in **CarlaSettingsDelegate.cpp**:

```
150   void UCarlaSettingsDelegate::CheckCarlaSettings(UWorld *world)
151   {
152     if (IsValid(CarlaSettings))
153     {
154       return;
155     }
156     if (world == nullptr || !IsValid(world) || world->IsPendingKill())
157     {
158       world = GetLocalWorld();
159     }
160     check(world != nullptr);
161     auto GameInstance  = Cast<UCarlaGameInstance>(world->GetGameInstance());
162     check(GameInstance != nullptr);
163     CarlaSettings = &GameInstance->GetCarlaSettings();
164     check(CarlaSettings != nullptr);
165   }
```

Any help is appreciated, as it is possible that my BSc thesis depends on this part. 😄 
Thank you in advance!

@bernatx @marcgpuig @Axel1092 @nsubiron

NnaYelsel, 2023-05-23:
Hi ! Just have the same problem with a custom optical flow mode, didn't change this piece of code either. 
I will try to find the rootcause. If you find it by yourself, please post the solution.

Best regards

NnaYelsel, 2023-05-23:
You can try to add #include "Carla/Game/CarlaGameInstance.h" on top of CarlaSettingsDelegate.cpp
That's worked for me, the project is compiling.

bernatx, 2023-05-23:
Hi,
That's a problem in windows where Unreal uses a unity way of putting together .h files and it is different for each compilation. So, this way there are .cpp that don't need to have the include to some .h and they work, because Unreal has put together several .h and the declaration of the objects is found by the .cpp. But some other times that is not the case (as in yours) and the .cpp compilation fails. When you add new files, the unity take another group of .h and then the compilation can fail.

In your case the **CarlaSettingsDelegate.cpp** file needs a header that was not included at version 0.9.14 that is **#include "Game/CarlaGameInstance.h"**. This line was included later on when we merge an option to not use the unity way of Unreal at compilation, and then each .cpp needs to have the .h that will use. You can disable unity using the ARGS="--no-unity", by the way.

In last version, we have added all the include sentences to all the .cpp, so it would always compile in unity or not unity way.

Regards

4doschris, 2023-05-23:
> Hi, That's a problem in windows where Unreal uses a unity way of putting together .h files and it is different for each compilation. So, this way there are .cpp that don't need to have the include [...]

Thanks a lot, it worked, the error codes are gone. Now I only have antoher one left, but that is another case.

closed 2023-05-23 by 4doschris