carla-simulator/carla #6497: Configuration Script Warning – libjpeg-turbo

Bojken-Noven, 2023-05-24:
**CARLA version**: 0.9.14
**Platform/OS**: Windows 11

I'm attempting to build the modified fork of Unreal Engine 4.26. When running GenerateProjectFiles.bat, all of the third party modules appear to build correctly except for libjpeg-turbo. This is the warning I get:

WARNING: Library 'C:\UE4\Engine\Source\ThirdParty\libjpeg-turbo\lib/Win64\turbojpeg.lib' was not resolvable to a file when used in Module 'LibJpegTurbo', assuming it is a filename and will search library paths for it. This is slow and dependency checking will not work for it. Please update reference to be fully qualified alternatively use PublicSystemLibraryPaths if you do intended to use this slow path to suppress this warning.

If I don't address this warning, compilation in VS Code 2019 fails since the necessary .lib file wasn't generated.

So far, I've tried identifying issues in the module's .Build.cs file without success. I understand that this is a common problem with the Unreal build system, but I'm unsure of how to solve it for this particular module.

hwwxj, 2023-08-14:
hi, do u find any solution to this issue now?

Bojken-Noven, 2023-08-30:
Nope, I gave up.