carla-simulator/carla #6498: Unable to create a Docker image containing a compiled version of Unreal Engine

ZubeenK, 2023-05-25:
Platform/OS: Linux (ubuntu 20.04)
Problem you have experienced: Failed to download 'http://cdn.unrealengine.com/dependencies/UnrealEngine-12372779-e1515af26c634d2a8ade60b1afd1f065/01bb78539fc8dda386d45f9b5615f9a1e8ca5d94' when trying to create docker image
Steps to reproduce:
Reference link to follow the step: https://github.com/carla-simulator/carla/blob/master/Util/Docker/README.md
I am getting  Failed to download 'http://cdn.unrealengine.com/dependencies/UnrealEngine-12372779-e1515af26c634d2a8ade60b1afd1f065/01bb78539fc8dda386d45f9b5615f9a1e8ca5d94' error, 
when I try to run the command: ue4-docker build 4.24.3 --no-engine --no-minimal
ue4-docker version: 0.0.98
Other information (documentation you consulted, workarounds you tried): Tried newer version of ue4-docker, changed the unreal engine version to 26.0, but still getting the same error.

[image]

Does anyone know how to resolve this issue?

chenhengwei1999, 2023-10-24:
Hi, I met this issue today. After I execute the command which build Unreal Engine, the errors would occur:

Command what I use:

```bash
ue4-docker build 4.24.3 --no-minimal
```

Termial's output:

```bash
Failed to download 'http://cdn.unrealengine.com/dependencies/UnrealEngine-18298425/0107275f6001142a756fd31da3aef73c3cf3e703': remote server returns Error: (403) Forbidden. (WebException)
```

According to [this question](https://forums.unrealengine.com/t/upcoming-disruption-of-service-impacting-unreal-engine-users-on-github/1155880) and [#issue 6486](https://github.com/carla-simulator/carla/issues/6486), there are troubles while access `cdn.unrealengine.com`. So, I am wandering if we can modify the `Dockerfile` under the `ue4-docker` command? After that, we can solve this problem. Have you any solutions for it, [ZubeenK](https://github.com/ZubeenK)?

muety, 2024-02-20:
Same problem here.