carla-simulator/carla #6513: Carla only using 1 CPU process. Is this normal?

MarioAlmelaUMH, 2023-05-30:
CARLA version: 0.9.12 (form source)
Platform/OS: Ubuntu 20.04.06 LTS
Problem you have experienced: 

Hello, I'm using Carla for a University research project and I'm noticing some performance issues. I've tried to profile them, and I got the results shown in this image:
[image]

As you can see, one of my CPU's processes caps to 100%, while my GPU usage is around 20-30%. My question is: is it possible to configure Carla so it uses more CPU processes instead of one? My GPU is being a bit underused, so is it possible for Carla to use more of the GPU's power?

Maybe my computer doesn't have the right specs to properly run Carla. My specs are:
- Memory: 62.5 GiB
- Processor: Intel® Core™ i9-9980XE CPU @ 3.00GHz × 36
- Graphics: NVIDIA Corporation GP104GL [Quadro P4000]

If there's a way to configure Carla so it uses more than a single CPU process, that will haste simulations by a lot, and make workflow more fluid. If it's an Unreal Engine problem, where should I look for answers?

Lesliewsq, 2023-09-13:
Same problem.Have you solved it?

MarioAlmelaUMH, 2023-09-13:
Sorry, but it's still the same. I have no clue as to why it happens.

Lesliewsq, 2023-09-13:
> Sorry, but it's still the same. I have no clue as to why it happens.

When I run this command, the following error appears. It seems that carla calls the integrated graphics card. But I can't locate the root of this problem. I added ‘-prefernvidia’ at the end of this command，but it didn't work.
`./CarlaUE4.sh -vulkan -RenderOffscreen -nosound 
4.26.2-0+++UE4+Release-4.26 522 0
Disabling core dumps.
../src/intel/isl/isl.c:2216: FINISHME: ../src/intel/isl/isl.c:isl_surf_supports_ccs: CCS for 3D textures is disabled, but a workaround is available.`

`./CarlaUE4.sh -vulkan -RenderOffscreen -nosound -prefernvidia
4.26.2-0+++UE4+Release-4.26 522 0
Disabling core dumps.
../src/intel/isl/isl.c:2216: FINISHME: ../src/intel/isl/isl.c:isl_surf_supports_ccs: CCS for 3D textures is disabled, but a workaround is available.`

Lesliewsq, 2023-09-13:
> Sorry, but it's still the same. I have no clue as to why it happens.

Also，there was a process of CARLA when I run `top`   but  `nvidia-smi`  has not. It seems that CARLA was unsuccessful to  calls the nvidia gpu.

MarioAlmelaUMH, 2023-09-21:
Did you install Carla from source? I did, but in some tutorials that use Carla (https://gitlab.com/hatem-darweesh-autoware.ai/carla-autoware/-/tree/master, https://autoware.org/running-multiple-autoware-powered-vehicles-in-carla-using-zenoh/) all of them are using the docker version (https://carla.readthedocs.io/en/0.9.14/start_quickstart/). Could that be the reason?

Lesliewsq, 2023-09-24:
I just downloaded the carla compressed package from the official website and then decompressed it. There was no problem starting up for a period of time, but one day suddenly this error persisted.

MarioAlmelaUMH, 2023-09-26:
Ok, I have downloaded the not-from-source version of Carla (from here: https://carla.readthedocs.io/en/0.9.14/start_quickstart/) and I have no more problems now. I don't know if in the future the issue will come back, but for now is as it should be: all cores working between 0 and 40% of usage. I don't need specifically the source version nor Unreal Engine right now, so for me the issue is solved.

closed 2023-09-26 by MarioAlmelaUMH

Morizb, 2024-04-07:
> > Sorry, but it's still the same. I have no clue as to why it happens.
> 
> Also，there was a process of CARLA when I run `top` but `nvidia-smi` has not. It seems that CARLA was unsuccessful to call the nvidia gpu.

I have also encountered this problem. Have you resolved it?