carla-simulator/carla #6518: UE4.26.2   carla 0.9.13   carsim 2023.0  co-simulation

CaoShuxing, 2023-05-31:
CARLA version: carla 0.9.13--build   UE4.26.2
Platform/OS: ubuntu 18.04
Problem you have experienced:
When I use manual_control_carsim.py, when I press the ctrl k key to activate my dynamics, the output file of carsim has the calculation results, but the vehicle in the virtual world does not move
What you expected to happen:
When I use manual_control_carsim.py, when I press the ctrl k button to activate my dynamics, the output file of carsim has the calculation results, and the vehicles in the virtual world also move accordingly

kidneyqsq, 2023-06-30:
Bro, how did you get vehicle dynamics plugin in ue4 ?
hxd, 车辆动力学插件怎么拿到的？

kidneyqsq, 2023-06-30:
I guess, you should press a key, like 'B' in the keyboard, to activate manual driving?
我感觉你应该按一个键来启动一下手动驾驶？

Lusikai, 2023-12-21:
I also want to ask this question, is there any way to obtain VehicleSim Dynamics Plugin?我也想问这个问题，VehicleSim Dynamics Plugin有什么办法可以获取吗？