carla-simulator/carla #6530: The UE4-CarlaUE4 Game has crashed and will close

yousef-sbeih, 2023-06-02:
Hi, I am facing an error while running the Carla simulator. The UE4-CarlaUE4 Game crashes and closes with a LowLevelFatalError [File:Unknown][Line: 136]. The exception 'close: An operation was attempted on something that is not a socket' is thrown.

Here are my device specifications:

Graphics Card: Nvidia 1660ti
RAM: 16GB
CPU: Intel Core i7 9th Gen
OS: Windows 11

Carla version: 0.9.14

The error occurs when I use the `reload_world ` method to reset the world and actors. It consistently happens at times and doesn't occur at other times. In some instances, the Carla simulator closes without displaying an error message.

hahne, 2023-06-05:
I see similar issues under Windows. E.g. when calling `toggle_camera()` in the CameraManager class of manual_control.py, it will also crash from time to time. It looks like it happens due to the `sensor.destroy()` call.

AlexanderInUM, 2023-07-09:
I'm having the same problem too, only after updating my Windows 11 to the latest version this morning. 
I tried CARLA_0.9.13 and CARLA_0.9.14, not working as it was before the update of latest version of Windows 11.

a19855352180, 2023-07-14:
I'm having the same problem too,How to solve it?

NeoBryant, 2023-09-17:
I'm having the same problem too. 
When I double-click the CarlaUE4 .exe file after the Windows 11 Version 22H2 system monthly update, I cannot start the carla server. The UE4-CarlaUE4 Game has crashed and will close.

sonalrpatel, 2023-09-20:
> I'm having the same problem too. When I double-click the CarlaUE4 .exe file after the Windows 11 Version 22H2 system monthly update, I cannot start the carla server. The UE4-CarlaUE4 Game has crashed and will close.

Even I am facing the same issue. Did you find any solution for it? then please let us know.

SExpert12, 2024-07-06:
Hi, 
what is that reload_world script? and How to use that. I am constantly getting that error.

dongfangyun, 2024-07-11:
i have seen this in CARLA 0.9.14 on Windows11 when i use manual_control.py:
i use "BACKSPCE" to restart() the world and change the car model randomly, Pressing the button "BACKSPCE" multiple times may result in this error.

SExpert12, 2024-07-11:
Okay. Thanks. I am using Ubuntu so I don't think it would work for me.

Thanks for reply.