carla-simulator/carla #6534: Mic -> Carla -> Zoom ?

Golddouble, 2023-06-04:
Hello

I am very new to Carla.

Is it possible with Carla to alienate voice my voice, before my voice goes into the video conferencing software "Zoom"?

If yes: What do I have to do?

Carla looks like this:
[image]
[image]

I guess I have to connect my microphone with a plugin. After I guess I would have to connect the plugin with Zoom. But I do not find a Zoom Box in Carla.

What are the correct steps?

Would appreciate some answer. Thank you.

Blyron, 2024-05-03:
I think you are in wrong carla repo, this is autonomus driving carla

Blyron, 2024-05-03:
A

closed 2024-05-03 by Blyron