carla-simulator/carla #6544: installation error make launch

k-shirakata-h, 2023-06-09:


raul-parada, 2023-11-08:
Is it solved? same issue here:

**********************************************************************
** Visual Studio 2019 Developer Command Prompt v16.11.31
** Copyright (c) 2021 Microsoft Corporation
**********************************************************************
[vcvarsall.bat] Environment initialized for: 'x64'

C:\Program Files (x86)\Microsoft Visual Studio\2019\Community>cd ..

C:\Program Files (x86)\Microsoft Visual Studio\2019>cd ..

C:\Program Files (x86)\Microsoft Visual Studio>cd ..

C:\Program Files (x86)>cd ..

C:\>cd carla

C:\carla>make PythonAPI
-[Setup]: [Batch params]: --boost-toolset msvc-14.2 --all
-[Setup]: Asynchronous jobs:  8
-[Setup]: Boost toolset:      msvc-14.2
-[Setup]: Install directory:  "C:\carla\Build\"
-[Setup]: Installing zlib...
    -[install_zlib]: [Batch params]: --build-dir "C:\carla\Build\"
    -[install_zlib]: Not downloading zlib because already exists the folder "C:\carla\Build\zlib-source\".
CMake Deprecation Warning at CMakeLists.txt:1 (cmake_minimum_required):
  Compatibility with CMake < 3.5 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

-- The C compiler identification is MSVC 19.29.30152.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - failed
-- Check for working C compiler: C:/Program Files (x86)/Microsoft Visual Studio/2019/Community/VC/Tools/MSVC/14.29.30133/bin/Hostx64/x64/cl.exe
-- Check for working C compiler: C:/Program Files (x86)/Microsoft Visual Studio/2019/Community/VC/Tools/MSVC/14.29.30133/bin/Hostx64/x64/cl.exe - broken
CMake Error at C:/Program Files/CMake/share/cmake-3.28/Modules/CMakeTestCCompiler.cmake:67 (message):
  The C compiler

    "C:/Program Files (x86)/Microsoft Visual Studio/2019/Community/VC/Tools/MSVC/14.29.30133/bin/Hostx64/x64/cl.exe"

  is not able to compile a simple test program.

  It fails with the following output:

    Change Dir: 'C:/carla/Build/zlib-source/build/CMakeFiles/CMakeScratch/TryCompile-4p5dw4'

    Run Build Command(s): "C:/Program Files/CMake/bin/cmake.exe" -E env VERBOSE=1 nmake -f Makefile /nologo cmTC_8bb16\fast
        C:\Program Files (x86)\GnuWin32\bin  -f CMakeFiles\cmTC_8bb16.dir\build.make /nologo -L                  CMakeFiles\cmTC_8bb16.dir\build
    'C:\Program' is not recognized as an internal or external command,
    operable program or batch file.
    NMAKE : fatal error U1077: 'C:\Program' : return code '0x1'
    Stop.

  CMake will not be able to correctly generate this project.
Call Stack (most recent call first):
  CMakeLists.txt:4 (project)

-- Configuring incomplete, errors occurred!

    -[install_zlib]: [CMAKE ERROR] An error ocurred while executing cmake command.
    -[install_zlib]: [CMAKE ERROR] Possible causes:
    -[install_zlib]:                - Make sure "CMake" is installed.
    -[install_zlib]:                - Make sure it is available on your Windows "path".
    -[install_zlib]:                - Make sure you have cmake 3.12.4 or higher installed.
    -[install_zlib]: Exiting with error...

-[Setup]:
   Ok, and error ocurred, don't panic!
   We have different platforms where you can find some help :)

   - Make sure you have read the documentation:
       http://carla.readthedocs.io/en/latest/how_to_build_on_windows/

   - If the problem persists, you can ask on our Github's "Building on Windows" issue:
       https://github.com/carla-simulator/carla/issues/21

   - Or just use our Discord channel!
       We'll be glad to help you there :)
       https://discord.gg/42KJdRj
make: *** [setup] Error 1

Yalda98, 2024-05-11:
Can anybody help solving this issue?