carla-simulator/carla #6545: Unable to use multi_GPU feature

carlyangdong, 2023-06-09:
CARLA version: 0.9.14
Platform/OS: Ubuntu 20.04(LTS)
Problem: Unable to use multi_GPU feature
Nvidia dirver version: 530.41.03

Hi，I tried to use the new multi-gpu feature on Carla. I have one server with two Nvidia 3090 GPUs.  
Using the nvidia-smi command, I can see four 3090 graphics cards，no process uses gpu, indicating that the driver is OK.
Refer this tutorial link: [carla-multi-gpu-feature](https://carla.readthedocs.io/en/latest/adv_multigpu/#carla-multi-gpu-feature)
1.start the primary server:
./CarlaUE4.sh -nullrhi
2.start the first rendering server:
./CarlaUE4.sh -carla-rpc-port=3000 -carla-primary-host=127.0.0.1 -ini:[/Script/Engine.RendererSettings]:r.GraphicsAdapter=0

The above execution is ok，and I run the python example also works fine. The server-side Carla server process uses the first GPU.

But when I start the second rendering server with the command below, the service fails to start.
./CarlaUE4.sh -carla-rpc-port=3001 -carla-primary-host=127.0.0.1 -ini:[/Script/Engine.RendererSettings]:r.GraphicsAdapter=1
The error is that the vulkan device cannot be found.
I modified GraphicsAdapter=1 to 2 or 3, the same error.

I found that /Script/Engine.RendererSettings is located at CARLA_0.9.14/CarlaUE4/Config/DefaultEngine.ini
```
[/Script/Engine.RendererSettings]
r.DefaultFeature.MotionBlur=True
r.BasePassOutputsVelocity=True
r.BasePassForceOutputsVelocity=False
r.AllowStaticLighting=False
r.DiscardUnusedQuality=True
r.DefaultFeature.Bloom=False
r.DefaultFeature.AmbientOcclusion=False
r.DefaultFeature.AmbientOcclusionStaticFraction=False
r.DefaultFeature.AutoExposure=False
r.CustomDepth=3
r.Streaming.PoolSize=4000
r.TextureStreaming=True
r.GenerateMeshDistanceFields=True
r.DistanceFieldBuild.EightBit=False
r.DistanceFieldBuild.Compress=False
r.DistanceFields.AtlasSizeXY=1024
r.DistanceFields.AtlasSizeZ=2048
r.DefaultFeature.AutoExposure.ExtendDefaultLuminanceRange=True
r.DefaultFeature.AntiAliasing=2
```
But no configuration related to r.GraphicsAdapter was found.
So, on a multi-GPU server, how can carla use multiple graphics cards to start multiple rendering server?
The most important thing is the r.GraphicsAdapter=N configuration, this should be the configuration of UE4, but I can't find the relevant information，can some one help me?

TurtleZhong, 2023-06-09:
use -RenderOffScreen from this[ post](https://github.com/carla-simulator/carla/issues/6543)

carlyangdong, 2023-06-12:
> use -RenderOffScreen from this[ post](https://github.com/carla-simulator/carla/issues/6543)

谢谢！通过-RenderOffScreen的确可以启动多个指定显卡的渲染服务了。
但是，实际测试，carla服务并没有均衡地使用多个显卡，仍然是某一个显卡使用率较高(40%左右)，其他基本为1%，clien的fps帧率也并没有多少提升。
有更详细的使用配置文档介绍么？