carla-simulator/carla #6547: type hints for carla pythonapi client

qhaas, 2023-06-10:
CARLA version: 0.9.14
Platform/OS: Ubuntu 22.04
Problem you have experienced: Lack of [type hints](https://docs.python.org/3.5/library/typing.html) when developing with the carla client PythonAPI.
What you expected to happen: Type hints are present in the carla client PythonAPI library and usable by IDEs (e.g. [pylance in vscode](https://github.com/microsoft/pylance-release)) or via command-line tooling (e.g. [pyright](https://microsoft.github.io/pyright/#/))
Steps to reproduce: Install the Python 3.5 wheel file, import the carla module into any sample code, add a few random functions from this module, and note that type hints are missing / unknown.
Other information (documentation you consulted, workarounds you tried):  This is a de facto re-opening of [Issue 1333](https://github.com/carla-simulator/carla/issues/1333) with a request for a more cross-platform (i.e. not exclusive to pycharm IDE) and standardized means to achieve type hints (i.e. using the now built-in functionality of newer Python releases).  I feel given the productivity benefits of type hints warrant it, along with [community interest](https://github.com/carla-simulator/carla/discussions/5755).  Type hints are described in [PEP-484](https://peps.python.org/pep-0484/) and were introduced with Python 3.5.  Standardization of packaging of type hints shipped with python 3.7 and is described in [PEP-561](https://peps.python.org/pep-0561/).  Given the carla client pythonapi relies on `boost::python` and underlying C++ based libraries / dependencies, it might not be possible to implement implicitly without [upstream feature additions](https://github.com/boostorg/python/issues/277).  Given CARLA 0.9.14 targets Python 3.5, an upgrade to Python 3.7+ might be needed in future releases to achieve the packaging of said hints.

xudooo, 2024-10-10:
Hi, did you solve the problem?  I installed CARLA client library in a virtual environment with command `pip3 install carla`, how do I make it possible to give method hints when using pythonapi?

qhaas, 2024-12-01:
As of carla 0.9.15, as installed from pypi with pip and python 3.9, no type hints.

KbKuuhaku, 2024-12-20:
I'm facing the same problem, no type hints and pyright is complaining everywhere inside carla-related project :(