carla-simulator/carla #6552: `get_random_location_from_navigation()` always returns `None` on Town12

vbrebion, 2023-06-13:
Hi!

As part of some ongoing tests, I am trying to use the new Town12 map.
However, I have noted that it is not possible to spawn any pedestrian on this map, as the `get_random_location_from_navigation()` function always returns `None`.

Would it be possible for you to take a look at this issue?

Best,
Vincent

---

**CARLA versions:**
0.9.14

**Platform/OS:**
Ubuntu 20.04

**Problem you have experienced:**
As described above, the default `get_random_location_from_navigation()` always returns `None` on Town12.

**Steps to reproduce:**
1. Launch the simulator
2. Load the Town12 map (using the `config.py` tool for instance)
3. Use the `generate_traffic.py` tool to spawn vehicles and pedestrians: 0 pedestrian is spawned, and if the value of `loc` (line 268 of the script) is printed, it will always be `None`

**Other information (documentation you consulted, workarounds you tried):**
I have looked through the documentation and the GitHub issues to find out if it was a deliberate choice or a known issue, but I couldn't find anything on this topic.

anaishalin, 2023-06-27:
Hello, 

I'm having exactly the same issue and haven't found a solution yet. 
In other maps, my code works fine to spawn pedestrians and make them walk. However, for a large map like Town12, I can't find a way to spawn pedestrians in the right location and make them walk. 

Does anyone know how to replace this `get_random_location_from_navigation()` that doesn't work for Town12? Or does anyone know a workaround? 

Thank you very much. 

Best, 

Anaïs

mawi42, 2023-06-30:
Hi, same for me, to spawn somehow some walker I used vehicle spawn points - which is of cause not nice...

anaishalin, 2023-07-03:
Hello, indeed I've tried that too, but then the pedestrians are on the road and you can't make them walk anyway. So it's not a solution for realistic simulation...

mawi42, 2023-07-03:
> Hello, indeed I've tried that too, but then the pedestrians are on the road and you can't make them walk anyway. So it's not a solution for realistic simulation...

Yes, you are right and I agree, it is not really a solution.

vbrebion, 2023-11-15:
For information, this issue is still present in version 0.9.15

ajperry2, 2023-11-18:
Could this be an issue with the navigation on the map not being present? I tried moving the xodr to the same folder as the map with no avail. Strange this seems to work in the demo they show for 9.14.

mawi42, 2024-02-01:
Did anybody find a solution?

AreopagX, 2024-02-09:
I'm having the same problem.

GoodarzMehr, 2024-02-28:
The function seems to work for me (Ubuntu 22.04, CARLA 0.9.15), though it seems to only return random locations with $|x| < 150$ and $|y| < 150$. This may be because those two large maps (Town12-13) don't have any navigation information, i.e. no Nav folder like the other ones. I'm trying to see if I can use [this guide](https://carla.readthedocs.io/en/latest/tuto_M_generate_pedestrian_navigation/) to generate that Nav information.

AreopagX, 2024-02-29:
@GoodarzMehr  You're right. I've also tried to use this guide to generate the navigation information, however, this did not really work out because of the way the Maps are created in the Unreal Editor (Or its just because I have no clue of UE).

Nevertheless, I found out that one can use the [OpenDrive Tutorial](https://carla.readthedocs.io/en/latest/adv_opendrive/) to generate the navigation information for maps 11, 12 and 13. For example, just run

`python3 /path/to/carla/PythonAPI/util/config.py -x /path/to/carla/CarlaUE4/Content/Carla/Maps/Town12/OpenDrive/Town12.xodr`

and the navigation file will be created in `/path/to/carla/CarlaUE4/Content/Carla/Maps/Nav/OpenDrive.bin`. Just rename it and move it to `/path/to/carla/CarlaUE4/Content/Carla/Maps/Town12/Nav/Town12.bin` and the pedestrian navigation should work.

Still, there are other problems like #7195 .

GoodarzMehr, 2024-02-29:
That's interesting, thanks for the info. I guess I've been encountering something similar in #7191.

GoodarzMehr, 2024-03-05:
@AreopagX I think I figured out how to generate pedestrian navigation information based on the tutorial [here](https://carla.readthedocs.io/en/latest/tuto_M_generate_pedestrian_navigation/), though I'm not sure if my end result is any different from yours. Here's what I did:

1. In UE4, for each tile select objects that have "\_Roads_" in their name. Then use **File** -> **Carla Exporter**. This should give you 36 (or 48 for Town 13) `.obj` files that describe the meshes relevant to pedestrian navigation.
2. Each `.obj` file uses its tile's local coordinate system, so the data needs to be transformed into the global coordinate system, which is described in the Town12 (or Town13) file. I used the code below to do the transformation.
```Python
import sys

# Check parameters.
if (len(sys.argv) < 5):
    print('Usage: original.obj transformed.obj x_transform y_transform')
    sys.exit(1)

try:
    f = open(sys.argv[1], 'rt')
    lines = f.readlines()
    f.close()

    d = open(sys.argv[2], 'wt')

    for line in lines:
        if line[:2] == 'v ':
            # Get vertex coordinates.
            split_line = line.split(' ')
            if not split_line:
                print(f'Problem parsing vertex coordinates: {line}')
                sys.exit(1)
            else:
                x = float(split_line[1]) + float(sys.argv[3])
                y = float(split_line[3]) + float(sys.argv[4])
                z = float(split_line[2])
                d.write('v %f %f %f\n' % (x, z, y))
            else:
                d.write(line)
	
    d.close()
except:
    print(f'Could not process {sys.argv[1]} file')
    sys.exit(1)
```
3. Now that all coordinates are transformed into the global coordinate system, use the `addOBJ.py` file under`Util/DockerUtils/dist` to add all tiles into one `Town12.obj` (or `Town13.obj`) file.
4. Now copy the `.xodr` file next to the created `.obj` file and use `./build.sh Town12`. The process takes about an hour or two to complete.

I have included the binaries I got here in case anyone wants to use them (The binary file is compressed twice, once into `.tar.xz` to get around GitHub's size limit, and then into `.zip` to get around the extension type limit).

[Town12.zip](https://github.com/carla-simulator/carla/files/14490338/Town12.zip)
[Town13.zip](https://github.com/carla-simulator/carla/files/14490340/Town13.zip)