carla-simulator/carla #6554: Problem about save_to_disk on windows

buaali, 2023-06-15:
CARLA version: 0.9.14
Platform/OS: windows 11
Problem you have experienced: When I use save_to_disk on ubuntu to save the image that the camera sensor listens to, there is no problem, but when I run the same code again on windows, I get a picture full of stripes, and the content of the picture is so damaged that I can't see it to any useful information. It is worth noting that I use the same computer to run ubuntu and windows dual systems, and the version of carla is the latest version. I tried using cv2 to save the image, but it didn't solve the problem.
What you expected to happen: Save pictures correctly on windows
Steps to reproduce:
Other information (documentation you consulted, workarounds you tried):rounds you tried):

VladMarianCimpeanu, 2023-07-27:
Hi, I had the same issue. By setting the camera parameters `image_size_x `= `image_size_y `= `k * 256`, where `k` is any integer, I could fix this problem.

TzabarDolev, 2023-09-19:
I wrote a short blog post about this issue on the Carla simulator research blog - https://carlasimblog.wordpress.com/2023/09/16/the-64-division-curse/. And added a short code snippet to fix this.

closed 2024-05-24 by Blyron