carla-simulator/carla #6556: Error while building Boost 1.70 - Make PythonAPI

ronyshaji, 2023-06-15:
I am facing an issue while buiding the libraries using PythonAPI:. The error is shown as below: Do you have any solution why the error is happening.

`        1 file(s) copied.
common.copy C:\pev-simulation-engine\Build\boost-1.72.0-install\boost-1_72\boost\compatibility\cpp_c_headers\cstring
C:\pev-simulation-engine\Build\boost-1.72.0-source\boost\compatibility\cpp_c_headers\cstring
        1 file(s) copied.
common.copy C:\pev-simulation-engine\Build\boost-1.72.0-install\boost-1_72\boost\compatibility\cpp_c_headers\cwctype
C:\pev-simulation-engine\Build\boost-1.72.0-source\boost\compatibility\cpp_c_headers\cwctype
        1 file(s) copied.
common.copy C:\pev-simulation-engine\Build\boost-1.72.0-install\lib\cmake\boost_filesystem-1.72.0\libboost_filesystem-variant-vc142-mt-x64-1_72-static.cmake
build\boost\bin.v2\libs\filesystem\build\msvc-14.2\release\address-model-64\link-static\threading-multi\libboost_filesystem-variant-vc142-mt-x64-1_72-static.cmake
        1 file(s) copied.
...failed updating 5 targets...
...skipped 4 targets...
...updated 15435 targets...
"Building failed at attempt number 1. Retrying..."
    -[install_boost]: Building...
Performing configuration checks

    - default address-model    : 32-bit (cached)
    - default architecture     : x86 (cached)
    - has_icu builds           : no  (cached)

Component configuration:

    - atomic                   : not building
    - chrono                   : not building
    - container                : not building
    - context                  : not building
    - contract                 : not building
    - coroutine                : not building
    - date_time                : building
    - exception                : not building
    - fiber                    : not building
    - filesystem               : building
    - graph                    : not building
    - graph_parallel           : not building
    - headers                  : not building
    - iostreams                : not building
    - locale                   : not building
    - log                      : not building
    - math                     : not building
    - mpi                      : not building
    - program_options          : not building
    - python                   : building
    - random                   : not building
    - regex                    : building
    - serialization            : not building
    - stacktrace               : not building
    - system                   : building
    - test                     : not building
    - thread                   : not building
    - timer                    : not building
    - type_erasure             : not building
    - wave                     : not building

...patience...
...patience...
...patience...
...found 34082 targets...
...updating 9 targets...
compile-c-c++ build\boost\bin.v2\libs\python\build\msvc-14.2\release\address-model-64\link-static\python-3.11\threading-multi\object\life_support.obj
life_support.cpp
libs\python\src\object\life_support.cpp(96): error C2106: '=': left operand must be l-value

    call "build\boost\bin.v2\standalone\msvc\msvc-14.2\address-model-64\architecture-x86\msvc-setup.bat"  >nul
 cl /Zm800 -nologo @"build\boost\bin.v2\libs\python\build\msvc-14.2\release\address-model-64\link-static\python-3.11\threading-multi\object\life_support.obj.rsp"

...failed compile-c-c++ build\boost\bin.v2\libs\python\build\msvc-14.2\release\address-model-64\link-static\python-3.11\threading-multi\object\life_support.obj...
compile-c-c++ build\boost\bin.v2\libs\python\build\msvc-14.2\release\address-model-64\link-static\python-3.11\threading-multi\object\class.obj
[... 483 lines cut ...]
-- Detecting CXX compile features
-- Detecting CXX compile features - done
CMake Error at CMakeLists.txt:4 (include):
  include could not find requested file:

    Build/CMakeLists.txt.in

-- Build debug:   ON
-- Build release: ON
-- Build test:    ON
CMake Error at LibCarla/cmake/server/CMakeLists.txt:11 (install):
  install FILES given directory "/$Recycle.Bin" to install.

-- Configuring incomplete, errors occurred!

-[BuildLibCarla]: [ERROR] An error ocurred while executing the cmake.
          [ERROR] Possible causes:
          [ERROR]  - Make sure "CMake" is installed.
          [ERROR]  - Make sure it is available on your Windows "path".
          [ERROR]  - CMake 3.9.0 or higher is required.
make: *** [LibCarla] Error 1

C:\pev-simulation-engine>

`

ronyshaji, 2023-06-15:
find the solution --> downgrade python to 3.8.10

roshambo919, 2023-10-06:
Newer versions of python will require newer versions of boost FYI.

yaogang2060, 2024-12-03:
> Newer versions of python will require newer versions of boost FYI.

boost-1.86.0 + python 3.12.7 on Windows 11 is fine.