carla-simulator/carla #6560: GNSS latitude and longitude format

JohnxLiang, 2023-06-16:
1.What is the GNSS output latitude and longitude format?
WGS84 is divided into the following three representation methods. Is Carla using one of them?

(1)Degree (°): N23.973875°, E120.982025° (23.973875°N, 120.982025°E)
(2)Degree-Minutes (° '): N23° 58.4325', E120° 58.9215' (23° 58.4325'N, 120° 58.9215'E)
(3)Degree-Minutes-Seconds (° ' "): N23° 58' 25.95", E120° 58' 55.29" (23° 58' 25.95"N, 120° 58' 55.29"E)

2.What does the plus sign stand for in latitude and longitude?

Is it true that in latitude, a plus sign represents north of the equator and a minus sign represents south of the equator? And in longitude, a plus sign represents east of the Prime Meridian (which passes through Greenwich, England) and a minus sign represents west of the Prime Meridian?

Thank you~^^

yjzhai-cs, 2024-01-25:
I had a similar problem. Have you solved the problem？

yjzhai-cs, 2024-01-25:
> 1.What is the GNSS output latitude and longitude format? WGS84 is divided into the following three representation methods. Is Carla using one of them?
>
> (1)Degree (°): N23.973875°, E120.982025° [...]

I had a similar problem. Have you solved the problem？