carla-simulator/carla #6561: carla ue4editor: blueprint runtime error, bp_Light.

xuq9908, 2023-06-16:
CARLA version:
0.9.13

Platform/OS:
Windows

Problem you have experienced:
The scene in UE4Editor looks overexposure.
[image]

and reports this error
Blueprint Runtime Error: "Accessed None trying to read property Weather". Bluprint: BP_Lights Function: Execute Ubergraph BP Lights
[image]

I tried to create a blank project, copied all contents and open the level there.
received this error:
The following member variables in blueprint BP_Weather have invalid type. Default Weathers
[image]

What you expected to happen:
The color of scene in editor looks normal as before.

Steps to reproduce:
I didn't change anything and suddenly received this error 3 days ago. It stuck at discovering assets, and then crash (even when opening a blank project)
[image]

I fixed this by adding these 2 lines into Config/DefaultEngine.ini:
[/Script/WindowsTargetPlatform.WindowsTargetSettings]
DefaultGraphicsRHI=DefaultGraphicsRHI_DX12

and after that, the whole scene becomes white as the above picture
Does anyone know how to turn it back to normal color? I tried re-install, but it didn't work. It seems to be the problem of sun intensity/ sky light intensity, but Idk why it's locked (in costum map), I'm not able to modify the value or to reset it to default value

xX-Conan-Xx, 2023-09-13:
Maybe considering replace the config folder with original one. That solved my problem

Blyron, 2024-05-03:
It seems your content folder binaries went corrupt  or sommething happened with them I would recommend to delete Unreal/CarlaUE4/Content/Carla folder and then download it again via Update.bat or cloning content repository in the same folder;
https://bitbucket.org/carla-simulator/carla-content/src/master/

closed 2024-05-24 by Blyron