carla-simulator/carla #6564: CARLAUE4 Engine Crashes

zehranrgi, 2023-06-18:
I installed carlaue4 and could open it. 
However, when I clicked the play button in it, it gives this error from terminal and is killed. Is there any way to solve it? 

```
[2023.06.18-00.12.45:724][ 38]LogShaderCompilers: Display: Worker (4/5): shaders left to compile 3544
[2023.06.18-00.12.45:724][ 38]LogSkeletalMeshReduction: Reducing skeletal mesh for LOD 1 
/home/../carla/Util/BuildTools/BuildCarlaUE4.sh: line 172: 10065 Killed                  ${GDB} ${UE4_ROOT}/Engine/Binaries/Linux/UE4Editor "${PWD}/CarlaUE4.uproject" ${RHI}
make: *** [Util/BuildTools/Linux.mk:7: launch] Error 137
```

Blyron, 2024-06-07:
Try again until it works, this is due to Vulkan being unstable. It eventually works. It happens with low VRAM GPUs. 
I am closing this issue if something comes up, feel free to reopen it.

closed 2024-06-07 by Blyron

HarrisonC7, 2024-07-20:
hei,I'd like to ask if you have solved this problem? thanks!