carla-simulator/carla #6565: Image vsync problem on visualize_multiple_sensors.py

funatsufumiya, 2023-06-19:
[image]

`visualize_multiple_sensors.py` showing like above, and I fixed this as workaround.

```python
    def save_rgb_image(self, image):
        t_start = self.timer.time()

        # print("width: ", image.width, " height: ", image.height) # 426 x 360

        pixel_size = image.width * image.height
        width = 448
        # width = image.width
        height = int(pixel_size / width)

        image.convert(carla.ColorConverter.Raw)
        array = np.frombuffer(image.raw_data, dtype=np.dtype("uint8"))
        array = np.reshape(array, (image.height, image.width, 4))
        array = array[:, :, :3]
        array = array[:, :, ::-1]

        # WORKAROUND: reshape and drop rows
        array = array.reshape((pixel_size, 3))
        array = array[:width*height, :].reshape((height, width, 3))

        if self.display_man.render_enabled():
            self.surface = pygame.surfarray.make_surface(array.swapaxes(0, 1))

        t_end = self.timer.time()
        self.time_processing += (t_end-t_start)
        self.tics_processing += 1
```

[image]

original code is here

```python
    def save_rgb_image(self, image):
        t_start = self.timer.time()

        image.convert(carla.ColorConverter.Raw)
        array = np.frombuffer(image.raw_data, dtype=np.dtype("uint8"))
        array = np.reshape(array, (image.height, image.width, 4))
        array = array[:, :, :3]
        array = array[:, :, ::-1]

        if self.display_man.render_enabled():
            self.surface = pygame.surfarray.make_surface(array.swapaxes(0, 1))

        t_end = self.timer.time()
        self.time_processing += (t_end-t_start)
        self.tics_processing += 1
```

I don't know why this is happening. Any helps?

CARLA version: 0.9.14
Platform/OS: Windows 11 (NVIDIA GeForce RTX3060), Python 3.7.16

funatsufumiya, 2023-06-19:
Main reason is probably related to `image_size_x` and `image_size_y` of BP.
https://github.com/carla-simulator/carla/blob/0.9.14/PythonAPI/examples/visualize_multiple_sensors.py#L114

I tried manually changing them into `1280x720`, no problems occured. ( but I don't know why... )

mawi42, 2023-06-30:
Hi,

this is because of GPU memory handling a stride issue, see also https://github.com/carla-simulator/carla/issues/6085

funatsufumiya, 2023-07-02:
@mawi42 Thank you. I close this issue as duplicated.

closed 2023-07-02 by funatsufumiya

TzabarDolev, 2023-09-18:
I just wrote a short blog post about this issue on the Carla simulator research blog - https://carlasimblog.wordpress.com/2023/09/16/the-64-division-curse/. And added a short code snippet to fix this.