carla-simulator/carla #6566: Secondary server returning None as data (Carla Multi-GPU Feature)

techner3, 2023-06-19:
CARLA version:0.9.14
Platform/OS: Ubuntu 20.04
Problem: Unable to use multi_GPU feature

Hi, I have started secondary server in different machine (local network). I was able to connect with the primary server. 
The Primary server run: ./CarlaUE4.sh --nullrhi
The secondary server run : ./CarlaUE4.sh -carla-rpc-port=4000 -carla-primary-host=127.0.0.1 -carla-primary-port=2002 -ini:[/Script/Engine.RendererSettings]:r.GraphicsAdapter=0

But during the simulation the secondary server is returning None as the data. Any idea how to solve it ?

Hoping someone could answer my question, thanks.

TurtleZhong, 2023-06-26:
Ref：[6545](https://github.com/carla-simulator/carla/issues/6545)

Blyron, 2024-05-03:
We are already having Ubuntu 22.04 support for UE4 
Look at build issues:
https://carla.readthedocs.io/en/latest/build_linux/