carla-simulator/carla #6572: How to get a 2D map contains both roads and buildings in Carla?

jeromefan, 2023-06-21:
CARLA version: 0.9.14 build from source, Platform/OS: Ubuntu 20.04.

Hi guys, is it possible that I can get a 2D map in Carla that contains both the roads and top-down shapes of buildings? Moreover, it would be awesome if the map is vectorized.

Python API `map.save_to_disk()` is good, but there is no shapes of the buildings. Looking forward to awesome guidance!

Blyron, 2024-05-03:
Currently we do not have that feature implemented we can create a request as a feature, but it won't coming back soon. What you can do is to create a camsera sensor and put it into the sky looking downwards. Other thing you can do is create a custom sensor which does it for you with ortographic camera. but yes, we do not have this feature. As this is not an issue I will be closing it. Are you happy with that?

qhaas, 2024-12-17:
This would be a useful feature for presentations / situational awareness of scenarios, planning to go the downward facing camera approach described by Blyron.

It would be easier to implement agnostic to map if we could programmatically get the usable (i.e. maximum bounding box where any vehicle could spawn on a road or walker spawn on a sidewalk) dimensions of the map to automatically place said camera for any given town.