carla-simulator/carla #6578: Connect Logitech G923

debipe20, 2023-06-22:
Hello,

I am planning to use a Logitech G923 Steering Wheel. Could anyone provide any instruction on how to setup Logitech Steering wheel with CARLA to run manual_control_steeringwheel.py?

CARLA version: 0.9.10
Platform/OS: Ubuntu 20.04

jimingda, 2023-06-26:
add a wheel_config.ini file

```
[G923 Racing Wheel]
steering_wheel = 0
throttle = 1
brake = 2
reverse = 5
handbrake = 4
```

and modify the  manual_control_steeringwheel.py

debipe20, 2023-11-07:
Got this error

Traceback (most recent call last):
  File "manual_control_steeringwheel.py", line 865, in <module>
    main()
  File "manual_control_steeringwheel.py", line 857, in main
    game_loop(args)
  File "manual_control_steeringwheel.py", line 787, in game_loop
    controller = DualControl(world, args.autopilot)
  File "manual_control_steeringwheel.py", line 229, in __init__
    self._joystick = pygame.joystick.Joystick(0)

jimingda, 2024-06-14:
Hi, @debipe20 

To check if a joystick is recognized by the system in Ubuntu, you can use the following command:  `dmesg | grep -i joystick`

For example, on my Ubuntu 20.04 system, I got the following result:

```
icub@icubsrv:~$ dmesg | grep -i joystick
[    4.919765] logitech 0003:046D:C266.0009: input,hidraw2: USB HID v1.11 Joystick [Logitech G923 Racing Wheel for PlayStation 4 and PC] on usb-0000:00:14.0-4/input0
```

If the joystick is recognized, you can test it with pygame:
```
import pygame

pygame.init()
pygame.joystick.init()

if pygame.joystick.get_count() == 0:
    raise Exception("No joystick connected")

joystick = pygame.joystick.Joystick(0)
joystick.init()
joystick_count = pygame.joystick.get_count()
for i in range(joystick_count):
    joystick = pygame.joystick.Joystick(i)
    joystick.init()
    print(f"Joystick {i} initialized: {joystick.get_name()}")
```
On my system, this script produces:
```
(carla) icub@icubsrv:/opt/carla-simulator/PythonAPI/examples$ python3 test.py 
pygame 2.4.0 (SDL 2.26.4, Python 3.7.16)
Hello from the pygame community. https://www.pygame.org/contribute.html
Joystick 0 initialized: Logitech G923 Racing Wheel for PlayStation 4 and PC
```

chaozheUB, 2025-01-16:
> [G923 Racing Wheel]
> steering_wheel = 0
> throttle = 1
> brake = 2
> reverse = 5
> handbrake = 4

Hi @jimingda  May I ask where should i put this wheel_config.ini file?
I am using 0.9.15 packaged version (https://github.com/carla-simulator/carla/releases)
and I am not sure about where to change this settings.

I found a wheel_config.ini example in the `Plugins\telecarla\telecarla_manual_control` folder but I don't adding the section there works as the `manual_control_steering_wheel.py` complained about `configparser.NoSectionError: No section: 'G29 Racing Wheel'` when parsing the wheel_config.ini. The parser seems to return empty value.

Thanks!