carla-simulator/carla #6591: How to run CARLA in no rendering mode in windows

prabanjan2419, 2023-06-27:
Dear Community,

I am trying to run Carla in No rendering mode in windows. I could able to run in low graphics mode by modifying the DefaultEngine.ini file in WindowsNoEditor\carlaUE4\Config. but I couldn't find what to modify in this file to run it in no rendering mode. could you please help?

[/Script/Engine.Engine]
bSmoothFrameRate=false
SmoothedFrameRateRange=(LowerBound=(Type="ERangeBoundTypes::Inclusive",Value=22),UpperBound=(Type="ERangeBoundTypes::Exclusive",Value=120))

[/Script/HardwareTargeting.HardwareTargetingSettings]
TargetedHardwareClass=Desktop
AppliedTargetedHardwareClass=Desktop
DefaultGraphicsPerformance=Maximum
AppliedDefaultGraphicsPerformance=Maximum

[/Script/EngineSettings.GameMapsSettings]
EditorStartupMap=/Game/Carla/Maps/Town10HD_Opt.Town10HD_Opt
GameDefaultMap=/Game/Carla/Maps/Town10HD_Opt.Town10HD_Opt
ServerDefaultMap=/Game/Carla/Maps/Town10HD_Opt.Town10HD_Opt
GlobalDefaultGameMode=/Game/Carla/Blueprints/Game/CarlaGameMode.CarlaGameMode_C
GameInstanceClass=/Script/Carla.CarlaGameInstance
TransitionMap=/Game/Carla/Maps/Town10HD_Opt.Town10HD_Opt
GlobalDefaultServerGameMode=/Game/Carla/Blueprints/Game/CarlaGameMode.CarlaGameMode_C

[/Script/Engine.RendererSettings]
r.DefaultFeature.MotionBlur=True
r.BasePassOutputsVelocity=True
r.BasePassForceOutputsVelocity=False
r.AllowStaticLighting=False
r.DiscardUnusedQuality=True
r.DefaultFeature.Bloom=False
r.DefaultFeature.AmbientOcclusion=False
r.DefaultFeature.AmbientOcclusionStaticFraction=False
r.DefaultFeature.AutoExposure=False
r.CustomDepth=3
r.Streaming.PoolSize=4000
r.TextureStreaming=True
r.GenerateMeshDistanceFields=True
r.DistanceFieldBuild.EightBit=False
r.DistanceFieldBuild.Compress=False
r.DistanceFields.AtlasSizeXY=1024
r.DistanceFields.AtlasSizeZ=2048
r.DefaultFeature.AutoExposure.ExtendDefaultLuminanceRange=True
r.DefaultFeature.AntiAliasing=2

chirudeep001, 2024-01-17:
In the command window:
you run using this command CarlaUE4 -NoRendering

NirobNabil, 2026-03-01:
> In the command window: you can run using this command CarlaUE4 -NoRendering

This didn;t work for me. @prabanjan2419 did you find a solution?