carla-simulator/carla #6597: make import/maps in Carla 0.9.13

AndritMuca, 2023-07-01:
CARLA version: 0.9.13
Platform/OS: Windows

Initially, I exported an OSM map from OpenStreetMap and attempted to convert it using the Carla script "osm to xodr.py". However, I encountered the following error consistently: Warning: The network contains extremely large coordinates, which may cause issues with Unreal Engine coordinates. It is recommended to limit any axis to a maximum value of 50000. In an attempt to address this problem, I tried converting the OSM file using the GitHub source "osmtoxodr" found at: https://github.com/JHMeusener/osm2xodr. Although this method worked, it only imported the roads and did not include traffic lights and other assets that I expected to be imported if the Carla Python script had functioned properly.

As an alternative approach, I exported the OSM file to FBX format using Blender. However, when I attempted to run the "make import" command, it was not successful. Nevertheless, I discovered that converting the OSM file to OBJ format and then utilizing an online converter, such as the one found at: https://products.aspose.app/3d/conversion/obj-to-fbx, seemed to work and allowed me to generate a functional FBX file, enabling me to create a map.

Then I proceeded with the "make launch" command, but encountered the following issues when opening the UMAP file:

[image]

At this point, I observed that there were no traffic lights or tables present on the streets, and when I attempted to run the "manual_control.py" script, the car was not found or sometimes even dident run the script.

I also attempted to manually connect the red points to the roads in an effort to address the issue directly at Unreal Engine, but unfortunately, it did not yield the desired results. It seems that there might be something missing beneath the map, such as a base or foundation, which is causing the car to fall through or not stay on the road properly.

I think that these problems may be related to the coordinates in the OSM file. Unfortunately, using Roadrunner for map editing or creation is not an option for me due to its lack of availability as a free tool. Consequently, I am seeking a solution that allows me to obtain the FBX and XODR files from the OSM map, enabling me to successfully import the map into Unreal Engine and create a scenario.

I would greatly appreciate any advice or suggestions you can provide to help resolve these issues. Thank you very much!

juanrodriguezdekra, 2024-02-05:
Hello! I'm facing the same issue as you, could you solve it? I'm stuck at this point and can't find an answer.

GladysKao, 2024-08-18:
Carla just can hold map smaller than 20km*20km