carla-simulator/carla #6600: osm_to_xodr.py not working

AndritMuca, 2023-07-03:
CARLA version: 0.9.13
Platform/OS: Windows

While converting the osm file from Openstreetmap to xodr with the script provided from Carla osm_to_xodr.py

python3 osm_to_xodr.py -i   path\map.osm -o  path\map.xodr

it says

Convert OpenStreetMap file to OpenDRIVE file. 
Warning: Discarding unusable type 'waterway.stream' (first occurence for edge '157911723').
Warning: Reducing junction cluster 1701386881,3508918959,880447302 (5 incoming edges).
Warning: Speed of straight connection '23761113_0->305473847_0' reduced by 19.28 due to turning radius of 17.76 (length=2.35, angle=23.26).      
Warning: Speed of straight connection '322857308#0_0->23761114_0' reduced by 5.99 due to turning radius of 47.91 (length=5.33, angle=21.58).     
Warning: Speed of straight connection '23758971_0->-322857308#1_0' reduced by 14.40 due to turning radius of 11.12 (length=4.35, angle=40.92).   
Warning: Speed of straight connection '871651449#0_0->322857308#0_0' reduced by 10.32 due to turning radius of 25.75 (length=14.26, angle=48.69).
Warning: Speed of straight connection '-39389724_0->-25307663#2_0' reduced by 5.47 due to turning radius of 90.49 (length=2.63, angle=16.69).
Warning: Network contains very large coordinates and could cause problems with Unreal Engine coordinates. The recomended maximum value in any axis is 50000
Warning: Could not compute smooth shape for edge '-60733475#3'.
Warning: Could not compute smooth shape from lane '227688325_0' to lane '23761113_0'. Use option 'junctions.scurve-stretch' or increase radius of junction '229946680' to fix this.
Warning: Could not compute smooth shape from lane '322857308#0_0' to lane '23761114_0'. Use option 'junctions.scurve-stretch' or increase radius of junction '256325620' to fix this.
Warning: 6 total messages of type: Speed of % connection '%' reduced by % due to turning radius of % (length=%, angle=%).

after that i run the xodr file to open it with the simulator and dont find the street. show nothing!

i open the xodr file with editor and the geometry values are very high:
[image]

where actually one example xodr file from Carla Town02 looks like these: 

[image]

and after opening the xodr file on simulator

[image]

I cant use Roadrunner as they dont answer and cant get a free version. I try also another converter from osm to xodr provided from: https://github.com/JHMeusener/osm2xodr and still not work well. 

Could anyone please help how to convert the osm to xodr so ica n build it then with the fbx file on unreal engine. 

Thank you

alanroyce2010, 2023-07-11:
i have the same problem. I'm trying to make a map of my collage but carla is throwing this output.
Converting OSM data to opendrive
pj_obj_create: Cannot find proj.db
pj_obj_create: Cannot find proj.db
pj_obj_create: Cannot find proj.db
Warning: Discarding unusable type 'waterway.drain' (first occurence for edge '30037017').
Warning: Discarding unusable type 'highway.corridor' (first occurence for edge '206412006').
Warning: Discarding unknown compound 'cycleway.shared_lane' in type 'cycleway.shared_lane|highway.cycleway' (first occurence for edge '447986442').
pj_obj_create: Cannot find proj.db
Warning: Speed of straight connection '25504237#2_0->206809952#0_0' reduced by 6.89 due to turning radius of 42.74 (length=18.55, angle=40.47).
Warning: Could not write OpenDRIVE geoReference. Only unshifted Coordinate systems are supported (center_map and use_offsets need to be set to False)
Warning: Could not compute smooth shape from lane '22688331-AddedOffRampEdge_2' to lane '24730725#0_0'. Use option 'junctions.scurve-stretch' or increase radius of junction '268790432' to fix this.
Warning: Could not compute smooth shape from lane '1039298143_1' to lane '28233862_1'. Use option 'junctions.scurve-stretch' or increase radius of junction '268790436' to fix this.
load opendrive map.
And it load an empty world. Can anyone help me with this?

andrejfsantos4, 2023-07-28:
I **don't** have a solution for this issue, but regarding some of those warnings:

> Warning: Network contains very large coordinates and could cause problems with Unreal Engine coordinates. The recomended maximum value in any axis is 50000

I believe this can be fixed with the option `center_map = True` in `carla.Osm2OdrSettings()`.

> i open the xodr file with editor and the geometry values are very high

This happened to me once and the result was just that the map was very far away from the origin in UE4, so I had to search around a bit until I found it.

> pj_obj_create: Cannot find proj.db

As mentioned [here](https://stackoverflow.com/questions/56764046/gdal-ogr2ogr-cannot-find-proj-db-error), this can be fixed by creating an environment variable PROJ_LIB that points to your proj.db, which in my installation was located at `carla_source\Build\proj-src\build\data`.

> Warning: Could not write OpenDRIVE geoReference. Only unshifted Coordinate systems are supported (center_map and use_offsets need to be set to False)

As mentioned [here](https://github.com/carla-simulator/carla/issues/3686#issuecomment-749446461), you can ignore this warning since OpenDRIVE geoReference is not used by CARLA.

Anhar2, 2024-05-03:
> As mentioned [here](https://stackoverflow.com/questions/56764046/gdal-ogr2ogr-cannot-find-proj-db-error), this can be fixed by creating an environment variable PROJ_LIB that points to your proj.db, [...]

Hi, I cannot find proj.db in the first place, is there something I should have installed first?