carla-simulator/carla #6613: Adding a 2 Wheeler in Carla 0.9.13

jemishGhoghari, 2023-07-05:
Hi,

I am facing a problem while adding a 2-wheeler. I am following the documentation provided by Carla. In the documentation, a reference skeleton is provided. So, I am confused about the binding of the bones. In the skeleton, We have a lot of bones.

So, My Question is, Which bones should be bonded with the 3D model of the 2-wheeler?
Additionally, There are 4 extra bones Wheel_FR, Wheel_FL, Wheel_RR, Wheel_RL. Should I bind these bones to the front wheel and rear wheel? 

Furthermore, I don't find "General2WheeledVehicleSkeleton" physics asset when importing the FBX model as written in the documentation. So, There, I am choosing "4WheeledBike" physics asset. In addition, After Importing, The Physics Asset looks like in the Images below.

[image]

Should It look like this? Because there is nothing in the Skeleton Tree. For other bikes which are already available in the Carla, like Harley, Vespa, etc., their skeleton tree structure looks like Bike_Rig, Wheel_FR, Wheel_FL, Wheel_RR, and Wheel_RL as bones added.

[image]

After All, was done, I registered the 2-wheeler blueprint as a vehicle in VehicleFactory blueprint. Then, I am trying to spawn my own 2-wheeler with Python Script "manual_control.py --filter myVehicle."  It does spawn the Vehicle, but the Vehicle does not move at all. 

In conclusion, What could be the clear solution to the Problem?

Regards

kshitiz-deloitte, 2023-08-21:
Hi @gsmario , I was trying to do the same (Adding 2-Wheeler in Carla).
I am able add the 2-wheeler and able to spawn in the same in manual control.
But the vehicle is acting unusual. (You can refer to the image below.)
[image]

Do you know what is the reason for this issue?

kshitiz-deloitte, 2023-08-31:
hi @jemishGhoghari, any update in this issue?

jemishGhoghari, 2023-08-31:
Hi @kshitiz-deloitte, Could you share the image of the PhysicsAsset of your Scooter? Maybe it's a problem with the Wheel Bone Setup!

kshitiz-deloitte, 2023-09-01:
> Hi @kshitiz-deloitte, Could you share the image of the PhysicsAsset of your Scooter? Maybe it's a problem with the Wheel Bone Setup!
[image]
 Here is the screenshot of the physics assets of the bike.

kshitiz-deloitte, 2023-09-01:
This is how it looks after spawning the bike
[image]

kshitiz-deloitte, 2023-09-04:
@jemishGhoghari 
I mistakenly uploaded the assets for bike, here is the physics asset for moped.
[image]

jemishGhoghari, 2024-06-26:
> Hi @gsmario , I was trying to do the same (Adding 2-Wheeler in Carla).
> 
> I am able add the 2-wheeler and able to spawn in manual control.
> 
> But the vehicle is acting unusual. (You can refer to the image below.)
> 
> [image]
> 
> 
> 
> Do you know what is the reason for this issue?

@kshitiz-deloitte I had the similar issues with Bike rider. But, Since, I am using it for VR purpose. My application did not required a Rider. So, I just removed it.

May be you can try to increase the sitting position of the rider.