carla-simulator/carla #6616: Getting "Map not found" for Town06, T0wn07 and Town12 in windows.

siddharthjain38, 2023-07-06:
**CARLA version:** 0.9.14
**Platform/OS:** Windows 10
**Problem you have experienced:** Getting "Map not found" for Town06, T0wn07 and Town12 in windows.
**What you expected to happen:** Load town 12 or additional maps
**Steps to reproduce**: Run below command. 
                                 `world = client.load_world("Town12")`

**Other information (documentation you consulted, workarounds you tried)**: As mentioned in the carla document I download AdditionalMaps and copied into carla root folder. Still I am not able to load additional Town.

[image]

viola521, 2023-07-12:
did you tackle this problem?

siddharthjain38, 2023-07-13:
> did you tackle this problem?
Hi Viola,

I was not able to use Town12 just by coping addition_maps in Carla root directory in windows. So I build Carla from source (https://carla.readthedocs.io/en/latest/build_windows/) using unreal-engine and than I am able to use Town12. 

By default Town06, Town07 and Town12 were not part of package so I change settings in unreal-engine to include these maps and after that I am able to access Town12.

MichelleSchaaf, 2023-08-04:
@siddharthjain38 do you use CARLA via the Unreal Editor, or have you generated your own Carla.exe? When I run CARLA in the Unreal Editor I have access to every map in the Unreal/CarlaUE4/Content/Carla/Maps directory, however in the exectuable I am not able to load any of those, even though client.get_available_maps() returns a whole bunch of them.

siddharthjain38, 2023-08-04:
@MichelleSchaaf I am able to access all available map and its working fine for me. Could you please share more details like Town No, what error you are getting etc.

MichelleSchaaf, 2023-08-04:
I have fixed my problem. Apparently it is because I created the executable with the Unreal Editor and not with **make package** #4632

kabdelmonem, 2024-02-23:
> I was not able to use Town12 just by coping addition_maps in Carla root directory in windows. So I build Carla from source (https://carla.readthedocs.io/en/latest/build_windows/) using unreal-engine [...]

hello i have the same problem how did you change the settings to include the three towns ?

NnaYelsel, 2024-02-27:
Hello all, I have the same problem, I really don't understand what is the solution for this problem ?

siddharthjain38, 2024-03-22:
@NnaYelsel @kabdelmonem To access these town you have to build carla and include  Town 12, 13, 15 in unreal engine before accessing. In unreal engine go to edit -> project setting and search list of maps. Here you have to add map that you wants to use.

Blyron-AeroSim, 2025-03-16:
https://carla.readthedocs.io/en/latest/core_map/#additional-maps