carla-simulator/carla #6617: make launch error 6 and 2

MAKER-park, 2023-07-07:
CARLA version: 0.9.14
Platform/OS: ubuntu 20.0.4
Problem you have experienced: 

`(base) park@park-Z97X-UD5H:~/Documents/carla$ make launch
Using clang-10 as the CARLA compiler.
Setup.sh: llvm-8.0 already installed.
Setup.sh: boost-1.80.0-c10 already installed.
Setup.sh: rpclib-v2.2.1_c5-c10 already installed.
Setup.sh: gtest-1.8.1-c10 already installed.
Setup.sh: recast-0b13b0-c10 already installed.
Setup.sh: Libpng already installed.
Setup.sh: Xerces-c already installed.
Setup.sh: Eigen already installed.
Setup.sh: Sqlite already installed.
Setup.sh: PROJ already installed.
Setup.sh: Patchelf already installed.
Setup.sh: CARLA version 0.9.14-6-g43b5e7064-dirty.
Setup.sh: Generating CMake configuration files.
Setup.sh: Success!
BuildLibCarla.sh: Building LibCarla "Server.Release" configuration.
ninja: no work to do.
[0/1] Install the project...
-- Install configuration: "Server"
BuildLibCarla.sh: Success!
BuildCarlaUE4.sh: Using Unreal Engine at '/home/park/UnrealEngine_4.26'
BuildCarlaUE4.sh: Build CarlaUE4 project.
make[1]: Entering directory '/home/park/Documents/carla/Unreal/CarlaUE4'
bash "/home/park/UnrealEngine_4.26/Engine/Build/BatchFiles/Linux/RunMono.sh" "/home/park/UnrealEngine_4.26/Engine/Binaries/DotNET/UnrealBuildTool.exe" CarlaUE4Editor Linux Development  -project="/home/park/Documents/carla/Unreal/CarlaUE4/CarlaUE4.uproject" 

Running Mono...

chmod: changing permissions of 'bin/mcs': Operation not permitted
chmod: changing permissions of 'bin/xbuild': Operation not permitted
Fixing inconsistent case in filenames.
Setting up Mono
~/UnrealEngine_4.26/Engine ~/Documents/carla/Unreal/CarlaUE4
Using 'git status' to determine working set for adaptive non-unity build (/home/park/UnrealEngine_4.26).
Creating makefile for CarlaUE4Editor (no existing makefile)
------- Build details --------
Using toolchain located at '/home/park/UnrealEngine_4.26/Engine/Extras/ThirdPartyNotUE/SDKs/HostLinux/Linux_x64/v17_clang-10.0.1-centos7/x86_64-unknown-linux-gnu'.
Using clang (/home/park/UnrealEngine_4.26/Engine/Extras/ThirdPartyNotUE/SDKs/HostLinux/Linux_x64/v17_clang-10.0.1-centos7/x86_64-unknown-linux-gnu/bin/clang++) version '10.0.1' (string), 10 (major), 0 (minor), 1 (patch)
Using bundled libc++ standard C++ library.
Using lld linker
Using llvm-ar : /home/park/UnrealEngine_4.26/Engine/Extras/ThirdPartyNotUE/SDKs/HostLinux/Linux_x64/v17_clang-10.0.1-centos7/x86_64-unknown-linux-gnu/bin/llvm-ar
Using fast way to relink  circularly dependent libraries (no FixDeps).
------------------------------
ERROR: Unhandled exception: System.UnauthorizedAccessException: Access to the path "/home/park/UnrealEngine_4.26/Engine/Intermediate/Build/Linux/B4D820EA/UE4Editor/Development/Link-libUE4Editor-Carla.so.link.sh" is denied.
         at System.IO.FileStream..ctor (System.String path, System.IO.FileMode mode, System.IO.FileAccess access, System.IO.FileShare share, System.Int32 bufferSize, System.Boolean anonymous, System.IO.FileOptions options) [0x0019e] in <98fac219bd4e453693d76fda7bd96ab0>:0 
         at System.IO.FileStream..ctor (System.String path, System.IO.FileMode mode, System.IO.FileAccess access, System.IO.FileShare share, System.Int32 bufferSize, System.IO.FileOptions options, System.String msgPath, System.Boolean bFromProxy, System.Boolean useLongPath, System.Boolean checkHost) [0x00000] in <98fac219bd4e453693d76fda7bd96ab0>:0 
         at (wrapper remoting-invoke-with-check) System.IO.FileStream..ctor(string,System.IO.FileMode,System.IO.FileAccess,System.IO.FileShare,int,System.IO.FileOptions,string,bool,bool,bool)
         at System.IO.StreamWriter.CreateFile (System.String path, System.Boolean append, System.Boolean checkHost) [0x0001c] in <98fac219bd4e453693d76fda7bd96ab0>:0 
         at System.IO.StreamWriter..ctor (System.String path, System.Boolean append, System.Text.Encoding encoding, System.Int32 bufferSize, System.Boolean checkHost) [0x00055] in <98fac219bd4e453693d76fda7bd96ab0>:0 
         at System.IO.StreamWriter..ctor (System.String path, System.Boolean append, System.Text.Encoding encoding, System.Int32 bufferSize) [0x00000] in <98fac219bd4e453693d76fda7bd96ab0>:0 
         at System.IO.StreamWriter..ctor (System.String path, System.Boolean append) [0x00008] in <98fac219bd4e453693d76fda7bd96ab0>:0 
         at (wrapper remoting-invoke-with-check) System.IO.StreamWriter..ctor(string,bool)
         at System.IO.File.CreateText (System.String path) [0x00000] in <98fac219bd4e453693d76fda7bd96ab0>:0 
         at UnrealBuildTool.LinuxToolChain.LinkFiles (UnrealBuildTool.LinkEnvironment LinkEnvironment, System.Boolean bBuildImportLibraryOnly, UnrealBuildTool.IActionGraphBuilder Graph) [0x009ab] in <ef6350ba47f74512ae03a5a567dbb6a0>:0 
         at UnrealBuildTool.UEToolChain.LinkAllFiles (UnrealBuildTool.LinkEnvironment LinkEnvironment, System.Boolean bBuildImportLibraryOnly, UnrealBuildTool.IActionGraphBuilder Graph) [0x00000] in <ef6350ba47f74512ae03a5a567dbb6a0>:0 
         at UnrealBuildTool.UEBuildBinary.Build (UnrealBuildTool.ReadOnlyTargetRules Target, UnrealBuildTool.UEToolChain ToolChain, UnrealBuildTool.CppCompileEnvironment CompileEnvironment, UnrealBuildTool.LinkEnvironment LinkEnvironment, System.Collections.Generic.List`1[T] SpecificFilesToCompile, UnrealBuildTool.ISourceFileWorkingSet WorkingSet, Tools.DotNETCommon.DirectoryReference ExeDir, UnrealBuildTool.IActionGraphBuilder Graph) [0x000ab] in <ef6350ba47f74512ae03a5a567dbb6a0>:0 
         at UnrealBuildTool.UEBuildTarget.Build (UnrealBuildTool.BuildConfiguration BuildConfiguration, UnrealBuildTool.ISourceFileWorkingSet WorkingSet, System.Boolean bIsAssemblingBuild, System.Collections.Generic.List`1[T] SpecificFilesToCompile) [0x00778] in <ef6350ba47f74512ae03a5a567dbb6a0>:0 
         at UnrealBuildTool.BuildMode.CreateMakefile (UnrealBuildTool.BuildConfiguration BuildConfiguration, UnrealBuildTool.TargetDescriptor TargetDescriptor, UnrealBuildTool.ISourceFileWorkingSet WorkingSet) [0x00146] in <ef6350ba47f74512ae03a5a567dbb6a0>:0 
         at UnrealBuildTool.BuildMode.Build (System.Collections.Generic.List`1[T] TargetDescriptors, UnrealBuildTool.BuildConfiguration BuildConfiguration, UnrealBuildTool.ISourceFileWorkingSet WorkingSet, UnrealBuildTool.BuildOptions Options, Tools.DotNETCommon.FileReference WriteOutdatedActionsFile, System.Boolean bSkipPreBuildTargets) [0x00012] in <ef6350ba47f74512ae03a5a567dbb6a0>:0 
         at UnrealBuildTool.BuildMode.Execute (Tools.DotNETCommon.CommandLineArguments Arguments) [0x0037d] in <ef6350ba47f74512ae03a5a567dbb6a0>:0 
         at UnrealBuildTool.UnrealBuildTool.Main (System.String[] ArgumentsArray) [0x002bb] in <ef6350ba47f74512ae03a5a567dbb6a0>:0
make[1]: *** [Makefile:315: CarlaUE4Editor] Error 6
make[1]: Leaving directory '/home/park/Documents/carla/Unreal/CarlaUE4'
make: *** [Util/BuildTools/Linux.mk:7: launch] Error 2
`

What you expected to happen:
https://carla.readthedocs.io/en/latest/build_linux/
just follow this site steps

try 
make clean -> make PythonAPI -> make launch 
same problem happened...

adrenalin7237, 2023-08-11:
I have the same problem as you... 
OS : ubuntu 20.04
carla : 0.9.14

Jackyuan994, 2023-10-19:
I have the same problem as you too...
OS: Ubuntu 20.04
carla : 0.9.14

hastaladisco, 2024-10-14:
i have the same problem. someone solved it?