carla-simulator/carla #6629: Distance between two locations in Carla simulator

SExpert12, 2023-07-12:
CARLA version:0.9.14
Platform/OS:Ubuntu
Problem you have experienced: Distance between two locations

I want to know is there any function in carla to check the distance between two locations? Suppose I have given two location carla.Location(x=-30.367612, y=7.337006,z=2.992777) and Location(x=2.376680, y=-3.337013, z=0.000000).
How to get the distance between two locations?

LucasFKobernic, 2023-07-12:
I thnik under carla.Location there is a method that calculates the distance between the given location and another one: https://carla.readthedocs.io/en/latest/python_api/

SExpert12, 2023-07-13:
Thank for your reply. It's spot on. Made my day.
Best wishes.

closed 2023-07-14 by SExpert12