carla-simulator/carla #6632: Setting up carla with unreal engine 4.26 on Linux, the git clone link for unreal engine not working. it requires username for github which :Support for password authentication was removed on August 13, 2021. And when I try to use the ssh link to clone it says: ERROR: Repository not found. fatal: Could not read from remote repository.  Please make sure you have the correct access rights and the repository exists.

Bo-Yu-Columbia, 2023-07-13:


skilh, 2023-07-13:
I have solved this issue with a personal access token. You can generate it and then use it as a password in this case.

siddharthjain38, 2023-07-14:
@Bo-Yu-Columbia Have you linked github account with unreal engine account. use - https://www.unrealengine.com/en-US/ue-on-github

Bo-Yu-Columbia, 2023-07-14:
yes i have

dscherrenburg, 2023-10-19:
I was running into the same issue just a minute ago. I got the same error messages both for following the tutorial and using my ssh-key. 
However, it is possible to clone unreal engine using your ssh-key. The mistake I made that lead to the same error (Please make sure you have the correct access rights and the repository exists.) was pressing enter to the following question: "Are you sure you want to continue connecting (yes/no/[fingerprint])?". After following the instructions by typing 'yes', the package cloned perfectly without any issues. 
So it is possible that you have made the same mistake and that the workaround would be just typing 'yes' to the question.

closed 2024-07-05 by Blyron

l-yerim23, 2025-04-04:
hi my name is yerim lee.
I like d