carla-simulator/carla #6636: instance segmentation camera

nolan4, 2023-07-14:
I am trying to use the instance segmentation camera in Carla 0.9.14 to determine what actors I should grab velocity and acceleration data from after each world.tick(). Here is what the docs say:

(https://carla.readthedocs.io/en/latest/ref_sensors/#instance-segmentation-camera)
"The server provides an image with the tag information encoded in the red channel: A pixel with a red value of x belongs to an object with tag x. The green and blue values of the pixel define the object's unique ID. For example a pixel with an 8 bit RGB value of [10, 20, 55] is a vehicle (Semantic tag 10) with a unique instance ID 20-55."

I verified that the cars I am looking at have the correct semantic tag in the red channel, but the values I get on the green and blue channels don't seem to correlate with their 'car.id'.

For a car actor which has a car.id = 317, my RGB vector is [10 (car label), 66, 81]. How do I connect GB values to car.id?

Gowtham-Raj-M, 2023-08-28:
@nolan4 , Did you find any solution to connect GB values to car.id?

alpi314, 2025-11-06:
Hey @Blyron , may I ask if Actor.GetUniqueID() is in any way related to CARLA's PythonAPI Actor.id or is it related to unreal engine ID?

In any case how could we relate the inverted id obtained from RGB colors to a specific CARLA python actor (for instance to only store bounding boxes of objects that are visible in camera, filtered by instance segmentation).

Thanks!

alpi314, 2025-11-06:
**CARLA 0.9.14, UE4**

When I spawn actors **(for this example only 1 vehicle),** I record their IDs as such:
```
vehicles = world.get_actors().filter('vehicle.*.*')
vehicles_ids = set([v.id for v in vehicles])
```

The resulting IDs are:
`{1479}`

However, when I decode the colors into unique IDs:
```
@staticmethod
def decode_instance_image(carla_image):
    a = np.frombuffer(carla_image.raw_data, dtype=np.uint8)
    a = a.reshape(carla_image.height, carla_image.width, 4)
    B = a[...,0].astype(np.uint16)
    G = a[...,1].astype(np.uint16)
    R = a[...,2].astype(np.uint8)
    semantic_tag = R
    instance_id = G + (B << 8)
    return semantic_tag, instance_id
```

The resulting unique codes are much higher:
23793

**(camera was aimed directly at the viechle to obtain only one code)**
Even with camera further away no code matches 1479.

Could my conversion be wrong?

alpi314, 2025-11-06:
Do you know in which version they are aligned?

Thank you!