carla-simulator/carla #6637: [CarlaUE4Editor] Error 6

SushilaRatre, 2023-07-14:
Hello All,

I am getting the same error after multiple reinstallations of CARLA while executing the command "make launch".
Please provide the solution for this. Thank you so much.

I am trying to install Carla 0.9.14 on windows 11 as per the instructions mentioned in the official document. 
[https://carla.readthedocs.io/en/latest/build_windows/]
Unearl Engine 4.26.2
Visual Studio 19

[image]
It is working properly till make pythonAPI. 

[image]

C:\carla\Unreal\CarlaUE4\Plugins\Carla\Source\Carla/Sensor/SceneCaptureSensor.h(15): fatal error C1083: Cannot open include file: 'Renderer/Public/GBufferView.h': No such file or directory
make: *** [CarlaUE4Editor] Error 6
!

Chang-byte, 2023-07-24:
I have encountered the same problem, have you solved it?
[image]

SushilaRatre, 2023-07-25:
Yes, I have successfully installed the Carla simulator after reinstallation
of the software.

SushilaRatre, 2023-07-25:
Hi! Try to follow the following link to solve the problem of fatal error.

https://www.drivereasy.com/knowledge/how-to-fix-unreal-engine-crash-quickly-and-easily/

SushilaRatre, 2023-07-25:
> I have encountered the same problem, have you solved it? [image]

Hi!, Try to use the following link to solve the problem of fatal error.
https://www.drivereasy.com/knowledge/how-to-fix-unreal-engine-crash-quickly-and-easily/

Chang-byte, 2023-07-26:
Thanks a lot,I'll try it

Blyron, 2024-10-18:
Those issues happens because u are trying to use a Vanilla version of Unreal instead of CARLA version. Please stick to build instructions.

closed 2024-10-18 by Blyron