carla-simulator/carla #6645: @AmirMz1 Radar will detect obstacles infront of it and this code shown will show the detected points with its velocity, azimuth, altitude and longitudinal distance from radar.

SExpert12, 2023-07-18:
@AmirMz1 Radar will detect obstacles infront of it and this code shown will show the detected points with its velocity, azimuth, altitude and longitudinal distance from radar.

You will need to do trial and error for mostly to know which points to take into consideration.

I think it is not possible to get the distance particularly from cars only. but my knowledge is limited.

You can see the images below for radar detection and its values

[image]
[image]
[image]
[image]

_Originally posted by @sbv198 in https://github.com/carla-simulator/carla/issues/5395#issuecomment-1221949743_
 Can anyone post the whole code. I am getting black list of points instead of values.
I want to use radar to measure the distance between moving cars.

SExpert12, 2023-07-19:
Radar sensor is not working in Imported map from RoadRunner. 
Can anyone help with this please?

sbv198, 2023-07-19:
It is better to use some algorithms to avoid unnecessary points detected by radar. 

Like in the image above with detected points it white points have the same velocity as ego while red and blue points have different  velocities so you can create some algorithms based on this  so it will eliminate all the points that have same velocities as ego

SExpert12, 2023-07-19:
But the problem is I am getting blank list of points. It is working for other town map but when I run the same program it shows blank list although two cars are moving opposite to each other. How to resolve this?
[image]