carla-simulator/carla #6653: Bug Report: Collision Not Detected When Spawning Actor with 'spawn' Function at Ego Vehicle's Location

GabrielZZZ, 2023-07-21:
**CARLA Version:** latest

**Platform/OS:** Ubuntu 20.04

**Problem Description:**

I have encountered a bug related to the `spawn` function in CARLA. When attempting to spawn an actor at a specific location (actor spawn location: Location(x=107.999817, y=132.417389, z=1.000000), Rotation(pitch=0.000000, yaw=49.000000, roll=0.000000)), it should have resulted in a collision with the ego vehicle (Location(x=107.999817, y=133, z=0.5)). According to the expected behavior, the actor should not have been spawned at this location to avoid the collision. However, in the CARLA simulator image, the actor has been spawned, and two vehicles crossed paths without any collision. The collision test report in the terminal also shows that no collision happened.

**Steps to Reproduce:**

1. Load CARLA simulator and set up a scenario with an ego vehicle at (Location(x=107.999817, y=133, z=0.5)).
2. Attempt to spawn another actor at (Location(x=107.999817, y=132.417389, z=1.000000), Rotation(pitch=0.000000, yaw=49.000000, roll=0.000000)) using the `spawn` function.
3. Observe that the actor is spawned successfully and crosses paths with the ego vehicle without any collision, despite the expected collision.

**Expected Behavior:**

The `spawn` function should prevent the actor from being spawned at a location where it would collide with the ego vehicle. It should detect possible collisions and avoid such spawning.

**Additional Information:**

- I have consulted the CARLA documentation and API reference for information on the `spawn` function and collision detection.
- I have tried using different spawn locations and rotations, but the issue persists.
- I have also tested the scenario with other actor types, and the collision detection seems to be working correctly for them.

Please let me know if you need any further information or if there are any workarounds to address this issue. Thank you for your assistance in resolving this bug.

GabrielZZZ, 2023-07-22:
Further testing, if we spawn a third actor following the same pattern, it will also cause models cross each other (i.e., bounding boxes cross?) while the no collision report given from CollisionTest.
[image]
[image]
[image]

GabrielZZZ, 2023-07-22:
To correct my expression, the CollisionTest has no report because I set the `second_actor.set_simulate_physics(enabled=False)`. If I set it to True, it will cause the collision, and the CollisionTest will give the collision report, which further means the spawn function should not allow the actors to be spawned at the given location.
[image]

[image]

SExpert12, 2023-07-22:
Hi, I am also trying to get collision information but it is not showing anything like your screenshot.
Can you please share your code to detect collision?

GabrielZZZ, 2023-07-24:
@SExpert12 I am using the built-in function CollisionTest from the scenario-runner of Carla. You find it here: https://github.com/carla-simulator/scenario_runner

SExpert12, 2023-07-25:
Okay. Can you please share your code if you are modifying anything in this. Because I don't want to write lines of code as I just need to attach collision sensor and get the time and location of vehicle when it occurs. I don't want to record all the collision events and other stuff.
Thanks for reply.

GabrielZZZ, 2023-07-25:
```
def _create_test_criteria(self):
        """
        A list of all test criteria will be created that is later used
        in parallel behavior tree.
        """
        criteria = []

        collision_criterion = CollisionTest(self.ego_vehicles[0])
        criteria.append(collision_criterion)

        return criteria
```

SExpert12, 2023-07-25:
Okay. Thank you so much.

GabrielZZZ, 2023-08-10:
We think the problem may exist in the judging process of bounding boxes cross on the z-coordinate. One reason is that the changes on either x or y coordinates do not trigger the bug. The other is that changing the model of the ego vehicle also triggered the bug, meaning the bounding boxes of the ego vehicle might not be the cause. 
 
[image]