carla-simulator/carla #6667: Unable to ingest OpenStreetMap into CARLA

andrejfsantos4, 2023-07-28:
**CARLA version:** 0.9.14 built from source
**Platform/OS:** Windows 10
**Problem you have experienced:** Unable to ingest a map from OpenStreetMap into CARLA. 

I've been trying to ingest a map exported from OpenStreetMaps into CARLA as per the [documentation](https://carla.readthedocs.io/en/latest/tuto_G_openstreetmap/), however I keep getting the error `map not found`. I have tried to ingest the map using all 3 methods described in the aforementioned documentation, but I always get the same error. There are no other errors. 

I'm able to load other maps (eg: Town01). I have also noticed that, when trying to ingest the .xodr/.osm file, the file 'carla_source/Unreal/CarlaUE4/Content/Carla/Maps/Nav/OpenDriveMap.obj' is modified and is non-empty.

**Steps to reproduce:** Run either
- `python config.py --osm-path=path_to_osm_file`
- `python config.py -x=path_to_xodr_file`

in the command line, with the CARLA server up and running.

**Other information (documentation you consulted, workarounds you tried):** I have tried to rebuild the CARLA server and UE4 (as per this issue #6006), but that didn't fix the issue. I have also tried using multiple different maps from OpenStreetMaps, both big and small. All other functionalities in CARLA appear to work fine.

Vafali, 2024-03-12:
Hi, I have tried on windows 11 it worked.
 
python config.py --osm-path="path\to\osm\file\your_file.osm"

I named my osm file as "map.osm". Anyway I will give you an example path to make it clear for you.

python config.py --osm-path="C:\Users\Admin\desktop\Carla_project\map.osm"

Blyron, 2024-05-31:
Hi!
Is this issue fix? Using absolute paths is always a good idea. 
Could you check if vafali 's solution work for you?