carla-simulator/carla #6681: fatal: repository 'https://github.com/CarlaUnreal/UnrealEngine.git/' not found

northbigpanda, 2023-08-01:
I tried to install Carla by the process of below link:
https://carla.readthedocs.io/en/0.9.14/build_linux/#unreal-engine,

when I proceeded to the step of installing UE4.26, i linked the UE account and Github account per the request of below instruction:
https://www.unrealengine.com/en-US/ue-on-github

then i continued with below command:
git clone --depth 1 -b carla https://github.com/CarlaUnreal/UnrealEngine.git ~/UnrealEngine_4.26

then an error emerged as below:
"remote: Repository not found.
fatal: repository 'https://github.com/CarlaUnreal/UnrealEngine.git/' not found",

so how to find the address of UE4.26, thanks a lot.

2019211753, 2023-08-02:
did u receive a email to conform the link? u can check it

northbigpanda, 2023-08-03:
> did u receive a email to conform the link? u can check it

yes, I received the mail and confirmed the link, still cannot git the clone…

2019211753, 2023-08-03:
emm..,do u have a organizations badge 'EPIC' in your github overview page?

northbigpanda, 2023-08-03:
> emm..,do u have a organizations badge 'EPIC' in your github overview page?

no, i cannot find it, could you send me a screenshot so that i can see what it is, thanks.

2019211753, 2023-08-03:
> > emm..,do u have a organizations badge 'EPIC' in your github overview page?
> 
> no, i cannot find it, could you send me a screenshot so that i can see what it is, thanks.

[image]

northbigpanda, 2023-08-04:
> > > emm..,do u have a organizations badge 'EPIC' in your github overview page?
> > 
> > no, i cannot find it, could you send me a screenshot so that i can see what it is, thanks.
> 
> [image]
> 
I don't have this badge, how did you get it?

2019211753, 2023-08-07:
> > > > emm..,do u have a organizations badge 'EPIC' in your github overview page?
> > > 
> > > 
> > > no, i cannot find it, could you send me a screenshot so that i can see what it is, thanks.
> > 
> > 
> > [image]
> 
> I don't have this badge, how did you get it?

sorry for my late reply. just follow the instructions step by step.

Blyron, 2024-04-26:
The clone link should be this one 
https://github.com/CarlaUnreal/UnrealEngine.git

Blyron, 2024-04-26:
And you have joined Epic Games organziation? 
If you have fixed this issue, are you happy to close this issue?

closed 2024-05-03 by Blyron

northbigpanda, 2024-05-11:
> And you have joined Epic Games organziation? If you have fixed this issue, are you happy to close this issue?

Yes, I have joined the Epic Games organination, thanks.

northbigpanda, 2024-05-15:
> And you have joined Epic Games organziation? If you have fixed this issue, are you happy to close this issue?

Yes, I'm happy to close this issue, thank you.