carla-simulator/carla #6685: ue4-docker build: error: unrecognized arguments: --no-engine

2019211753, 2023-08-02:
I follow the [instruction](https://github.com/carla-simulator/carla/tree/master/Util/Docker) to Ingestion in a CARLA package.
After i execuate the code 'ue4-docker setup', i get the following message:
```bash
usage: ue4-docker build [-h] [--ue-version UE_VERSION] [--linux] [--rebuild] [--dry-run] [--no-minimal] [--no-full]
                        [--no-cache] [--target TARGET] [--random-memory] [--exclude {ddc,debug,templates}] [--opt OPT]
                        [--cuda VERSION] [--visual-studio {2017,2019,2022}] [-username USERNAME] [-password PASSWORD]
                        [-repo REPO] [-branch BRANCH] [-isolation ISOLATION] [-basetag BASETAG] [-suffix SUFFIX] [-m M]
                        [-ue4cli UE4CLI] [-conan-ue4cli CONAN_UE4CLI] [-layout LAYOUT] [--combine] [--monitor]
                        [-interval INTERVAL] [--ignore-blacklist] [-v] [-changelist CHANGELIST]
                        [--prerequisites-dockerfile PREREQUISITES_DOCKERFILE]
                        [release]
ue4-docker build: error: unrecognized arguments: --no-engine
```
How can i solve it?

chenhengwei1999, 2023-10-23:
hello, I met this issue too, have you solved it?

2019211753, 2023-10-26:
> hello, I met this issue too, have you solved it?

no, i use another way to building carla from source

chenhengwei1999, 2023-10-26:
Hi, could you share another way or method to build carla from source? I would be very glad to hear from you.

2019211753, 2023-10-28:
> Hi, could you share another way or method to build carla from source? I would be very glad to hear from you.

I follow this instruction [Linux build](https://carla.readthedocs.io/en/latest/build_linux/) instead of docker build.

YSXXXXXXX, 2024-08-08:
Hi, I met the same problem, and I found this [pull request](https://github.com/adamrehn/ue4-docker/pull/273). If I understand it correctly, the ue4-docker does not build the ue4-engine by default.