carla-simulator/carla #6699: Carla's self-built vehicle is normal in ue4editor, but it is stuck in the ground when running in carla

AGREATAPE, 2023-08-08:
Hello CARLA community,
I built my own car in blender according to the official tutorial (the model is relatively small, less than 1m in size), and then imported it into ue4editor. If you drag the blueprint of the built model directly into ue4editor and click play to run, you can see When the car landed normally, there was no mold penetration
[image]
But if I set up my car in VehicleFactory and use manual_control.py to select my self-built car, the car will sink into the ground, as shown in the figure below:
[image]
[image]
Does anyone know what's going on, I've been stuck for several days

Alexued, 2023-11-24:
Same problem. Is that fixed?

AGREATAPE, 2023-11-25:
> Same problem. Is that fixed?

I didn't find the cause of the problem, but no such problem occurred after using a larger tire diameter

silencial, 2024-02-06:
> > Same problem. Is that fixed?
> 
> I didn't find the cause of the problem, but no such problem occurred after using a larger tire diameter

Also encountered this problem, and from the issues seems like many other did, too.
Do you think using larger tire diameter will cause other problems? Like the dynamics might be different?