carla-simulator/carla #6706: XerxesC CMake Error: (missing: XercesC_VERSION)

CoReNeli, 2023-08-10:
Platform/OS: Windows 11
Python: 3.8.10
CARLA version: 0.9.13 (with manually updated zlib verion to 1.2.13 in the carla/Util/InstallersWin/install_zlib.bat) full depth

I ran into problems with XercesC when running make PythonAPI. After fixing the XercesC Version from 3.2.3 to 3.2.4 at the following places

carla/Util/BuildTools/Setup.sh (Line 428) XERCESC_VERSION = 3.2.4
carla/Util/BuildTools/Setup.bat (Line 195) ...\xerces-c-3.2.4-install\lib ...
carla/Util/InstallersWin/install_xercesc.bat  (Line 43) set XERCESC_VERSION=3.2.4

I get the XercesC source and install folders unter carla\Build with the right version, 
```
-[install_xercesc]: xerces has been successfully installed in "C:\CarlaDReyeVR\carla\Build\xerces-c-3.2.4-install\"!
    -[install_xercesc]: Exiting...
        1 Datei(en) kopiert.
```

but I still get the following error:

```
CMake Error at C:/Program Files/CMake/share/cmake-3.27/Modules/FindPackageHandleStandardArgs.cmake:230 (message):
  Failed to find XercesC (missing: XercesC_VERSION)
Call Stack (most recent call first):
  C:/Program Files/CMake/share/cmake-3.27/Modules/FindPackageHandleStandardArgs.cmake:600 (_FPHSA_FAILURE_MESSAGE)
  C:/Program Files/CMake/share/cmake-3.27/Modules/FindXercesC.cmake:112 (FIND_PACKAGE_HANDLE_STANDARD_ARGS)
  CMakeLists.txt:71 (find_package)
```

Please help. I am at my wits end after checking almost every other CMake or Xercesc file within carla for a possible cause over the past days.

b1llow99, 2023-08-16:
This happened to me too, waiting for someone to help with the answer.

CoReNeli, 2023-08-16:
Please let me know if you have found another way in the meantime.

Looking at the call stack I tried the CMakeLists.txt directly under the carla directory but that only has 3 lines. The CMakeLists.txt under the carla\build\xerces-c-3.2.4-source directory has no input in line 71, so those didn't bring any other possible solution into view for me. I will try to ping someone on the Carlasim discord server....

Axel1092, 2023-08-17:
Hi @CoReNeli, aparently you have an error while building the XercesC library. Even if you get the  `-[install_xercesc]: xerces has been successfully installed...` something seems to be wrong, so I suggest you delete any XercesC folder inside the Build folder, run again the command and share the log with us so that we may take a look.

CoReNeli, 2023-08-18:
Many thanks!
I will attach the log in a txt file. I'm using the x64 Native Tools Command Prompt for VS 2019.
[Carla_makePythonAPI_currentlog_coreneli.txt](https://github.com/carla-simulator/carla/files/12377180/Carla_makePythonAPI_currentlog_coreneli.txt)

And it did show shortly at a seperate loading screen that a xercesc .zip was being unwrapped.

Axel1092, 2023-08-18:
Ok, I see now the issue. You need to edit the BuildOSM2ODR.bat to set the XercesC version change, 
```
        -DXercesC_INCLUDE_DIR=%INSTALLATION_DIR:/=\%\xerces-c-3.2.3-install\include^
        -DXercesC_LIBRARY=%INSTALLATION_DIR:/=\%\xerces-c-3.2.3-install\lib\xerces-c.lib^
```
you need to set up these two lines to use 3.2.4.

There is another error in the log, the `error: invalid command 'bdist_wheel'` which is fixed with `pip install wheel`.
Hope this helps.

CoReNeli, 2023-08-29:
This solved the problem! Thank you very much!