carla-simulator/carla #6712: Lavapipe is not a conformant vulkan implementation, testing use only

jingyue01, 2023-08-14:
CARLA version: 0.9.14

Platform/OS: ubuntu20.04

Problem you have experienced: On a same computer, I can run Carla on Windows, but it will flash back on Ubuntu with the following error code:
**./CarlaUE4.sh
4.26.2-0+++UE4+Release-4.26 522 0
Disabling core dumps.
WARNING: Lavapipe is not a conformant vulkan implementation, testing use only
XIO:  fatal IO error 13 (Permission denied) on X server ":0"
      after 293 requests (293 known processed) with 24 events remaining.
terminating with uncaught exception of type std::__1::system_error: mutex lock failed: Invalid argument
Signal 6 caught.
Segmentation fault (core dumped)**

What you expected to happen: Carla succesfully runs.

Steps to reproduce: run on VMWare, Amd gpu 5600xt. I have installed vulkan, but I don't know what is lavapipe and the IO error.

matthew-rococo, 2023-09-08:
The problem is mostly the nvidia gpu driver problem.  you can run 'nvidia-smi'  to check wether the gpu driver is ok.

closed 2023-10-08 by jingyue01

WebsterTien, 2024-09-16:
Can you tell me how you solve this problem?