carla-simulator/carla #6713: Capability of cameras capturing images at the same time

mgao97, 2023-08-16:


MichelleSchaaf, 2023-08-17:
Hey,

check out this example script: [sensor_synchronization.py](https://github.com/carla-simulator/carla/blob/master/PythonAPI/examples/sensor_synchronization.py). It uses a queue in the sensor callback. I have adapted the sensor_callback method, so it saves the generated images:
```python
    def sensor_callback(self, sensor_data, sensor_queue, sensor_name,cc):
        # Do stuff with the sensor_data data like save it to disk
        # Then you just need to add to the queue
        save_path = os.path.join(self._args.outputImg, sensor_name, str(sensor_data.frame)+".png")
        sensor_queue.put((sensor_data.save_to_disk(save_path, cc), sensor_data.frame))
```
you then only need to call this method when your camera "listens":
```python
cam.listen(lambda data: self.sensor_callback(data, sensor_queue, sensor_name, carla.ColorConverter.Raw))
```

mgao97, 2023-08-17:
Thank you for your message!

I also need to record the positions and rotations of cameras during capturing each image, could this be achieved by the sensor_callback? Another question is about the relationship between the setting of 30 fps and the driving speeds of vehicles. 

Thank you so much!

Blyron, 2024-05-03:
You can get each camera transform on each frame so yeah. FPS determines simulation speed. No driving speeds

Blyron, 2024-05-10:
Are you happy to close this issue with the answer given?

closed 2024-05-12 by mgao97