carla-simulator/carla #6719: BUG: Problem with get_matrix() method of carla.Transform Class

iaoqian, 2023-08-17:
CARLA version: 0.9.13
Platform/OS: Windows 10

I was working on merging multiple point cloud collected by Depth Camera and something else which both of them require the coordinates transformation between Carla World Coordinates System <-> Carla Camera Coord-Sys <-> 2D Pixel Coord-Sys.
I would demonstrate the problem I experienced in transforming coordinates in Camera Coord-Sys to World (that will be same in world to camera, cuz this procedure is invertible). 
According to [#553 Transform](https://github.com/carla-simulator/carla/issues/553), we know that we need swap axis when transforming coordinates from Camera Coord-Sys to  World by left multiplying a matrix:
[image]
and then, in theory, all I should do is left multiplying a matrix that rotates and translate coordinates originally in Camera Coord-Sys to World, and we can use get_transform().get_matrix() method of some carla.Actor (like a camera here) in Carla to get that matrix and make things running. But it actually made something going wrong.
### The following image shows tow point cloud computed by Depth and RGB camera 
(the implementation of *raw_data_to_depth* and *depth_to_local_point_cloud* is from [here](https://github.com/carla-simulator/driving-benchmarks/blob/master/version084/carla/image_converter.py), which is used in [official example](https://github.com/melfm/carla/blob/master/PythonClient/point_cloud_example.py) as well).
#### black point cloud is captured 20 Frames after point cloud that has true color (ignore BGR and RGB problem here) and you can find the car was turning left.
[image]
So I checked the code in [carla/LibCarla/scource/geom/Transform.h](https://github.com/carla-simulator/carla/blob/43b5e7064872bb6a9529664c2218e29df38dca04/LibCarla/source/carla/geom/Transform.h#L23), find:
```cpp
    /// Computes the 4-matrix form of the transformation
    std::array<float, 16> GetMatrix() const {
      const float yaw = rotation.yaw;
      const float cy = std::cos(Math::ToRadians(yaw));
      const float sy = std::sin(Math::ToRadians(yaw));

      const float roll = rotation.roll;
      const float cr = std::cos(Math::ToRadians(roll));
      const float sr = std::sin(Math::ToRadians(roll));

      const float pitch = rotation.pitch;
      const float cp = std::cos(Math::ToRadians(pitch));
      const float sp = std::sin(Math::ToRadians(pitch));

      std::array<float, 16> transform = {
          cp * cy, cy * sp * sr - sy * cr, -cy * sp * cr - sy * sr, location.x,
          cp * sy, sy * sp * sr + cy * cr, -sy * sp * cr + cy * sr, location.y,
          sp, -cp * sr, cp * cr, location.z,
          0.0, 0.0, 0.0, 1.0};

      return transform;
    }
```
But, in theory, we should rotate and translate coordinates with following arguments $(x, y, z, -yaw, -pitch, -roll)$,  where yaw, pitch, roll is the rotation of camera relative to world coordinates system. Precisely, we should do:
[image]
By comparing the formula, we can find that what the code in Carla implementation is actually transforming with arguments $(x, y, z, yaw, -pitch, -roll)$
It can explain why the those point cloud misaligned in the yaw level. So I implemented those part of code on my own and found it worked at some point. but there are still some confusing things.
[image]
### Finally, I find out that setting y axis of camera to -y can solve this problem. It kind of makes sense, cuz we set y coordinates to -y when swapping axis, but I have no idea to prove its correctness.
but anyway, I eventually got the perfect result in this way and I didn't really see some works like this (those guys work on it probably met some problem like [#3051 Merging multiple point cloud from depth camera](https://github.com/carla-simulator/carla/issues/3051) | [#6435 Merge/align point clouds from different perspectives](https://github.com/carla-simulator/carla/issues/6435)
[image]
### I could be wrong, but these things convinced me that it might be a BUG in Carla.  
### As for my work, including code & demo video, It' s literally available [here](https://github.com/iaoqian/carla_3d-mapping_and_adding_virtual_objects).

Blyron, 2025-04-01:
It is not a  bug due to how UE handles coordinates. Good Job by the way!

kkaiwwana, 2025-04-03:
Hi@Blyron , It's not very clear to me that what do mean by "how UE handles coordinates". If you're referring to UE's axis systems, differentiating from screen space as described in #553, then everyone knows it and correctly apply it. Or you actually mean the cpp code 'GetMatrix()' is actaully a standard widly used in UE related stuffs, I'm just kind of confused why it transforms target by (x, y, z, yaw, -pitch, -roll). 

And the fact is it has truly caused many related issues. If it's not a bug, it means there is a "rigth way" to use official api `get_transform()` making everything right. As far as I tried (~aug 2023), there were no means to make that happen until I use my implemented transformation. And my implementation indeed helps solving related issues. In this way, there are two possible causes, the first one, both offical implementation and mine are correct, where the former requires more operations, which is still unknow to us, to make things right outside the function `get_transform()` or it should not cause any issue. The other one is, the offical implementation is incorrect.

Would you like to provide some explaination or insights about these concerns?

Blyron, 2025-04-03:
Hello! This might be my assumption. Sorry for this.  I guess what GetMatrix() does, is return the transformation Matrix within unreal space. That's what I meant. 

Making a transform to right handed you have to apply rotation matrix, not just inverting the axis. And maybe is true that you need to do some more operations. 

https://forums.unrealengine.com/t/ue4-coordinate-system-not-right-handed/80398/4

We would be happy to get a PR with a new method for our Transform class named; GetRightHandedTransformMatrix on our Transform class.

kkaiwwana, 2025-04-03:
Thanks for the reply. I will try to figure out what exactly happened with these coordination and axises stuffs. If we can find the ultimate cause of these issues and inconsistent phenomenon, I’m glad to sent a pr to Carla.

zhoujiawei3, 2025-04-07:
It seems that when I use get_matrix(), the rotation matrix I get is in right handed coordinate

Blyron, 2025-04-08:
Hi @kkaiwwana 

First of all. Thank you for your effort we will be glad to receive an PR from you so your work is reflected in CARLA. This get_matrix was created before I joined CARLA so you may be right. Once you open the PR I will ask around in the team to see if it's like it. 

Of course we are open to suggestions and PRs to make life easier for our users. You can open issues with "feature request template"

kkaiwwana, 2025-04-08:
@Blyron hi, I opened a feature request issue here #8819.