carla-simulator/carla #6730: How to run multi-gpu feature in the source version of CARLA?

AGREATAPE, 2023-08-24:
CARLA version:0.9.14
Platform/OS:Ubuntu 20.04
Problem you have experienced: How to run multi-gpu feature in the source version of CARLA

Hi，I tried to use the new multi-gpu feature on Carla. I have one server with two Nvidia 2080ti and find this tutorial [carla-multi-gpu-feature](https://carla.readthedocs.io/en/latest/adv_multigpu/#carla-multi-gpu-feature); 
However, since I'm using the source version of CARLA and the corresponding files ./CarlaUE4.sh are not available (this is also explained in[ this link](https://carla.readthedocs.io/en/latest/build_faq/#carlaue4sh-script-does-not-appear-when-downloading-from-github)), how can I enable multi-GPU functionality in the source version of CARLA?

 I can't find the relevant information，can some one help me?

veinlove55, 2024-03-27:
Hello @AGREATAPE , have found the way to run multi gpu by the source CARLA?

Blyron, 2024-04-19:
You will need to create a package and then try it. For this you can do a "make package".

Blyron, 2024-04-26:
I am closing this issue, feel free to reopen if something comes up

closed 2024-04-26 by Blyron