carla-simulator/carla #6731: package 'carla_pointcloud' not found

wenjing1170, 2023-08-24:
CARLA version: 0.9.13(Debian)
Platform/OS: Ubuntu 20.04 LTS, RTX3080, nvidia driver 535, cuda 11.6, ROS2 galactic, autoware.universe
Problem you have experienced:

I follow the official guidence to run [openplanner](https://github.com/ZATiTech/open_planner) demo  on  carla and autoware.autoware paltform. When I run the demo, the error is as follow:

[image]
[image]

There is no 'carla_pointcloud' dkpg in my system environment. I have no idea. Thanks.

TianYifan-Vincent, 2024-05-15:
@wenjing1170 Sorry to bother you, I am experiencing the same problem when running open_planner(ROS2 humble), can you share your solution to this problem?

TianYifan-Vincent, 2024-05-15:
> It's been so long that I can't remember the exact solution. It seems that the file path of my source code is wrong. It seems that openplanner must be placed in the appropriate path. You can try placing autoware, openplaner, and carlabridge strictly according to the path in the openplanner tutorial.
> […](#)

> It's been so long that I can't remember the exact solution. It seems that the file path of my source code is wrong. It seems that openplanner must be placed in the appropriate path. You can try placing autoware, openplaner, and carlabridge strictly according to the path in the openplanner tutorial.
> […](#)

Thank you your reply, but i have a question: do i need to rebuild the project? Because I indeed didn't find the package  'carla_pointcloud'  in folder autoware/install as the problem raised.  Is the existence of this package relevant?