carla-simulator/carla #6741: Accident detector in CARLA

AOOOOOA, 2023-08-30:
Hi,

I notice there is a collision detector in Carla sensor used for detecting the collision event. But I haven't found the corresponding code block for the collision detector implementation. Could you please tell me where I can find it? 
And I'm also curious about whether Carla has other accident detectors besides the collision detector. Could you please help me with that? @krmartin @sunpochin @Jazzinghen @cesarsouza @armando-fandango 
Thanks!

TzabarDolev, 2023-09-18:
Hi, 
I just wrote a short blog post about collision detectors on the Carla simulator research blog - https://carlasimblog.wordpress.com/2023/09/18/collision-detection-systems-in-cars-enhancing-safety-and-realism-in-carla-simulator/, and added a code example for it.

AOOOOOA, 2023-09-19:
> Hi, I just wrote a short blog post about collision detectors on the Carla simulator research blog - https://carlasimblog.wordpress.com/2023/09/18/collision-detection-systems-in-cars-enhancing-safety-and-realism-in-carla-simulator/, and added a code example for it.

Thanks so much. Your blog is clear and helps a lot. But I want to know how the collision sensor works in Carla. i.e. how does the collision sensor know there is a collision happened? I have searched the code but didn't find the corresponding code snippet.

TzabarDolev, 2023-09-19:
> Thanks so much. Your blog is clear and helps a lot. But I want to know how the collision sensor works in Carla. i.e. how does the collision sensor know there is a collision happened? I have searched [...]

Hi, Actually it was answered here - https://github.com/carla-simulator/carla/issues/4409

AOOOOOA, 2023-09-20:
> Hi, Actually it was answered here - #4409

Thank U so much!

Blyron, 2024-04-26:
Hi
We are closing this issue, please reopen again if something comes up

closed 2024-04-26 by Blyron