carla-simulator/carla #6744: Unable to build UnrealEngine_4.26

FinCreWorld, 2023-08-30:
I'm building UnrealEngine_4.26, but when run './Setup.sh', I encountered download failure.
[image]

or by https
[image]

But I can download the file from the url, so how can I download the file by `Setup.sh` or put already downloaded file in the local cache?

Thanks so much !

FinCreWorld, 2023-08-30:
The network error is still not fixed. But I tried to identify the exact location of the cache file. And then I put the downloaded file in the folder manually.
[image]

Yin-xiongjie, 2023-12-01:
Hello friends, I met the same problem, do you know the cause of the problem? And how did you solve it? If I download the file manually, where should I put it?

j-cyoung, 2023-12-01:
> Hello friends, I met the same problem, do you know the cause of the problem? And how did you solve it? If I download the file manually, where should I put it?
[image]

I don't know the cause of this problem. But I tried to get the location of the file by checking update time of all related folders, and then I found the cache folder shown in the image above. You can see this folder (`.git/ue4-gitdeps/c3`) sharing the same prefix with the missing file (`c33ef...`). I downloaded the file manually and put it intot he folder.

Blyron, 2024-04-19:
Please if the issues are related with Unreal Engine, please refer to Epic Games forum. 
If you follow carla build that should be working

Blyron, 2024-04-26:
I am closing this issue, feel free to reopen if something comes up

closed 2024-04-26 by Blyron