carla-simulator/carla #6745: CarlaUE4/Plugins/StreetMap/Source/StreetMapRuntime/Private/StreetMapComponent.cpp:15:10: fatal error: 'Carla/BlueprintLibary/MapGenFunctionLibrary.h' file not found

aidenyao, 2023-08-30:
CARLA version:0.9.14
Platform/OS:ubuntu18.04.5
Unreal Engine Version: 4.26.2
Problem you have experienced:
I want to use StreetMap unreal plugins (https://github.com/carla-simulator/StreetMap),but when i make launch, it show errors:
carla-0.9.14/Unreal/CarlaUE4/Plugins/StreetMap/Source/StreetMapRuntime/Private/StreetMapComponent.cpp:15:10: fatal error: 'Carla/BlueprintLibary/MapGenFunctionLibrary.h' file not found。How to solve it?

zhaozhiqing1, 2023-11-24:
请问一下这个问题你解决了吗

Blyron, 2024-04-26:
That Plugin depends completely on CARLA so is installed with carla when you make a make launch. Please, try to use it with CARLA itself. if you do not want to, please refer to original plugin https://github.com/ue4plugins/StreetMap

closed 2024-04-26 by Blyron

Blyron, 2025-06-26:
> I solved this issue ,carla/Unreal/CarlaUE4/Plugins, I clone streetmap in
> this dir, just delete it , dont clone any package manually

No, we are not supporting 4.27 in short term

Aaron Samaniego Diaz