carla-simulator/carla #6750: Build Carla on Jetson or release the docker using on Jetson

CNWindson, 2023-09-04:
CARLA version: 
Any

Platform/OS: 
Jetson tx2 
(system of ubuntu 18.04.6
architecture of linux/arm64/v8)

Problem you have experienced:
I have searched the whole internet without the solution of using Carla on Jetson platform. I think it's really powerful and low energy consumption. I have tried using docker on Jetson but failed. I haven't tried to build it from source because I can't find any success
examples.

What you expected to happen:
1. Could you plz give me some advice about how to build from source on jeston platform? Is that worth to try?
2. Could you release the docker that can be used on Jetson? That will be really useful to do some work like 24 *7 data collection, algorithms reproduction and so on with a really low energy consumption. ( I have device of 1080Ti with power of 250W  and Jetson tx2 of only 7.5W) 

I'm really looking forward of your reply.

ahmedyossef788, 2023-12-13:
Were you able to run carla on jetson?

CNWindson, 2024-01-04:
I haven't 🥲