carla-simulator/carla #6758: Can not control multiple vehicle by carla-ros-bridge.

gigohe2, 2023-09-07:
CARLA version: 0.9.13

Platform/OS: ubuntu 20.04

Problem you have experienced:
 I want to control two vehicle by carla-ros-bridge carla_manual_control.launch .
So, I edited my launch file which spawns new car for control. 
/carla-ros-bridge/catkin_ws/src/ros-bridge/carla_spawn_objects/config/objects.json
I added "type": "vehicle.ford.crown",
            "id": "juncar", ~~ sensor lists~~~
to that json format file.

and then, I edited launch file of carla_manual_control node.
Inside the directory (/carla-ros-bridge/catkin_ws/src/ros-bridge/carla_manual_control/launch)
There is a file named 'carla_manual_control.launch'.
So, I copied and pasted it to same directory by naming 'custom_carla_manual_control.launch'
And I edited this file like it.
<launch>

  <node pkg="carla_manual_control" type="carla_manual_control.py" name="ego_manual_control" ns="ego_vehicle" output="screen">
    <param name="role_name" value="ego_vehicle"/>
  </node>

  <node pkg="carla_manual_control" type="carla_manual_control.py" name="juncar_manual_control" ns="juncar" output="screen">
    <param name="role_name" value="juncar"/>
  </node>
</launch>

I use 3 commands to bring it up.

1. roslaunch carla_ros_bridge carla_ros_bridge.launch

2. roslaunch carla_spawn_objects carla_spawn_objects.launch

3. roslaunch carla_manual_control custom_carla_manual_control.launch

When i launch my custom carla manual control launch file, two pygame windows come up.

[image]

And also, there are juncar topics.
[image]

When i published throttle topic to juncar, It didn't work.
The yellow ford car doesn't move.
How can i solve it?

What you expected to happen:
Because there are available topics to juncar, I think that i can control juncar by publishing topics. But it didn't work.

Please help me.
Thank you

vatch123, 2024-06-10:
Do you have any updates on this? I am stuck on the same issue

HMX2013, 2025-04-12:
> Do you have any updates on this? I am stuck on the same issue

Have you solved this? It seems that you have to modify bridge.py to add the controller of another car, but I failed.

gigohe2, 2025-04-18:
In the carla_ros_bridge.launch file, 

[image].

So, need to match the role name of each car following above launch file