carla-simulator/carla #6760: Carla 0.9.13 'make PythonAPI' fails on Windows 11

StijnOosterlinck, 2023-09-08:
Hi everyone, I'm trying to install Carla on Windows 11 but I am running into some problems. When trying to compile the PythonAPI using "make PythonAPI', the build fails but with no specific error message. I'll add the console output as a file since it is quiet large, but probably only the end is relevant. Has anyone experienced this issue before? 
Some details:
Carla 0.9.13
Visual studio 2019

This is the relevant part of the console output (I think):

C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.29.30133\bin\HostX86\x64\cl.exe /c /nologo /Ox /W3 /GL /DNDEBUG /MD -Idependencies/include -IC:\Python\Python311\include -IC:\Python\Python311\Include -IC:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.29.30133\ATLMFC\include -IC:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.29.30133\include -IC:\Program Files (x86)\Windows Kits\NETFXSDK\4.8\include\um -IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\ucrt -IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\shared -IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\um -IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\winrt -IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\cppwinrt -IC:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.29.30133\ATLMFC\include -IC:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.29.30133\include -IC:\Program Files (x86)\Windows Kits\NETFXSDK\4.8\include\um -IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\ucrt -IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\shared -IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\um -IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\winrt -IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\cppwinrt /EHsc /Tpsource/libcarla/libcarla.cpp /Fobuild\temp.win-amd64-3.11\Release\source/libcarla/libcarla.obj /experimental:external /external:W0 /external:I dependencies/include/system /DBOOST_ALL_NO_LIB /DBOOST_PYTHON_STATIC_LIB /DBOOST_ERROR_CODE_HEADER_ONLY /D_WIN32_WINNT=0x0600 /DHAVE_SNPRINTF /DLIBCARLA_WITH_PYTHON_SUPPORT -DLIBCARLA_IMAGE_WITH_PNG_SUPPORT=true /MD
libcarla.cpp
dependencies/include/system\boost/python/object/make_instance.hpp(50): error C2106: '=': left operand must be l-value
dependencies/include/system\boost/python/object/class_wrapper.hpp(42): note: see reference to function template instantiation 'PyObject *boost::python::objects::make_instance_impl<T,Holder,boost::python::objects::make_ptr_instance<T,Holder>>::execute<Src>(Arg &)' being compiled
        with
        [
            T=carla::client::Actor,
            Holder=boost::python::objects::pointer_holder<boost::shared_ptr<carla::client::Actor>,carla::client::Actor>,
            Src=boost::shared_ptr<carla::client::Actor>,
            Arg=boost::shared_ptr<carla::client::Actor>
        ]
dependencies/include/system\boost/python/object/class_wrapper.hpp(42): note: see reference to function template instantiation 'PyObject *boost::python::objects::make_instance_impl<T,Holder,boost::python::objects::make_ptr_instance<T,Holder>>::execute<Src>(Arg &)' being compiled
        with
        [
            T=carla::client::Actor,
            Holder=boost::python::objects::pointer_holder<boost::shared_ptr<carla::client::Actor>,carla::client::Actor>,
            Src=boost::shared_ptr<carla::client::Actor>,
            Arg=boost::shared_ptr<carla::client::Actor>
        ]
dependencies/include/system\boost/python/object/class_wrapper.hpp(41): note: while compiling class template member function 'PyObject *boost::python::objects::class_value_wrapper<boost::shared_ptr<carla::client::Actor>,boost::python::objects::make_ptr_instance<T2,boost::python::objects::pointer_holder<boost::shared_ptr<carla::client::Actor>,carla::client::Actor>>>::convert(Src)'
        with
        [
            T2=carla::client::Actor,
            Src=boost::shared_ptr<carla::client::Actor>
        ]
dependencies/include/system\boost/python/converter/as_to_python_function.hpp(27): note: see reference to function template instantiation 'PyObject *boost::python::objects::class_value_wrapper<boost::shared_ptr<carla::client::Actor>,boost::python::objects::make_ptr_instance<T2,boost::python::objects::pointer_holder<boost::shared_ptr<carla::client::Actor>,carla::client::Actor>>>::convert(Src)' being compiled
        with
        [
            T2=carla::client::Actor,
            Src=boost::shared_ptr<carla::client::Actor>
        ]
dependencies/include/system\boost/python/object/class_metadata.hpp(254): note: see reference to class template instantiation 'boost::python::objects::class_value_wrapper<boost::shared_ptr<carla::client::Actor>,boost::python::objects::make_ptr_instance<T2,boost::python::objects::pointer_holder<boost::shared_ptr<carla::client::Actor>,carla::client::Actor>>>' being compiled
        with
        [
            T2=carla::client::Actor
        ]
dependencies/include/system\boost/python/object/class_metadata.hpp(230): note: see reference to function template instantiation 'void boost::python::objects::class_metadata<W,X1,X2,X3>::maybe_register_pointer_to_python<T2>(T2 *,boost::mpl::false_ *,boost::mpl::false_ *)' being compiled
        with
        [
            W=carla::client::Actor,
            X1=boost::noncopyable,
            X2=boost::shared_ptr<carla::client::Actor>,
            X3=boost::python::detail::not_specified,
            T2=carla::client::Actor
        ]
dependencies/include/system\boost/python/object/class_metadata.hpp(229): note: see reference to function template instantiation 'void boost::python::objects::class_metadata<W,X1,X2,X3>::maybe_register_pointer_to_python<T2>(T2 *,boost::mpl::false_ *,boost::mpl::false_ *)' being compiled
        with
        [
            W=carla::client::Actor,
            X1=boost::noncopyable,
            X2=boost::shared_ptr<carla::client::Actor>,
            X3=boost::python::detail::not_specified,
            T2=carla::client::Actor
        ]
dependencies/include/system\boost/python/object/class_metadata.hpp(218): note: see reference to function template instantiation 'void boost::python::objects::class_metadata<W,X1,X2,X3>::register_aux2<T,use_callback>(T2 *,Callback)' being compiled
        with
        [
            W=carla::client::Actor,
            X1=boost::noncopyable,
            X2=boost::shared_ptr<carla::client::Actor>,
            X3=boost::python::detail::not_specified,
            T=carla::client::Actor,
            T2=carla::client::Actor,
[... 18 lines cut ...]
            X2=boost::shared_ptr<carla::client::Actor>,
            X3=boost::python::detail::not_specified
        ]
dependencies/include/system\boost/python/object/class_metadata.hpp(204): note: see reference to function template instantiation 'void boost::python::objects::class_metadata<W,X1,X2,X3>::register_aux(void *)' being compiled
        with
        [
            W=carla::client::Actor,
            X1=boost::noncopyable,
            X2=boost::shared_ptr<carla::client::Actor>,
            X3=boost::python::detail::not_specified
        ]
dependencies/include/system\boost/python/class.hpp(170): note: see reference to class template instantiation 'boost::python::objects::class_metadata<W,X1,X2,X3>' being compiled
        with
        [
            W=carla::client::Actor,
            X1=boost::noncopyable,
            X2=boost::shared_ptr<carla::client::Actor>,
            X3=boost::python::detail::not_specified
        ]
dependencies/include/system\boost/python/class.hpp(589): note: see reference to class template instantiation 'boost::python::class_<carla::client::Actor,boost::noncopyable,boost::shared_ptr<carla::client::Actor>,boost::python::detail::not_specified>::id_vector' being compiled
dependencies/include/system\boost/python/class.hpp(589): note: while compiling class template member function 'boost::python::class_<carla::client::Actor,boost::noncopyable,boost::shared_ptr<carla::client::Actor>,boost::python::detail::not_specified>::class_(const char *,boost::python::no_init_t)'
C:\joan\carla\PythonAPI\carla\source\libcarla\Actor.cpp(129): note: see reference to function template instantiation 'boost::python::class_<carla::client::Actor,boost::noncopyable,boost::shared_ptr<carla::client::Actor>,boost::python::detail::not_specified>::class_(const char *,boost::python::no_init_t)' being compiled
C:\joan\carla\PythonAPI\carla\source\libcarla\Actor.cpp(92): note: see reference to class template instantiation 'boost::python::class_<carla::client::Actor,boost::noncopyable,boost::shared_ptr<carla::client::Actor>,boost::python::detail::not_specified>' being compiled

-[BuildPythonAPI]: Carla lib for python has been successfully installed in "C:\joan\carla\PythonAPI\carla\dist"!

yijiaguo02, 2023-09-30:
seems I have the same problem

StijnOosterlinck, 2023-09-30:
Seems like the problem for me is that when compiling, Python 3.11 is used, instead of the python 3.7.2 installed in the active conda environment. I don't know why this is happening because I execute 'make PythonAPI' after 'conda activate carla7', which is the conda environment with python 3.7.2 installed...

level1coder, 2024-10-23:
① You need to block the original python environment;
② You need to change 'py -3 setup.py bdist_egg bdist_wheel' in carla\Util\BuildTools\BuildPythonAPI.bat to 'py - 3.7 ...';
③ You need to install the wheel for the current python version;
Then, you can generate it successfully!